gnuradio/gnuradio #925: vector_source_c throws 'std::bad_alloc' when used with tags

mbr0wn, 2016-09-16:
Stumbled over this while trying to implement a new QA.

'gnuradio-config-info -v': 3.7.10

A minimal "failing" example is attached.

<pre>
src = blocks.vector_source_c(data, (tag, ))
# src = blocks.vector_source_c(data)
</pre>

the active line here will throw an error after waiting for a few seconds:
<pre>
thread[thread-per-block[1]: <block vector_sink_c (2)>]: std::bad_alloc
</pre>

The version without tags does work just fine.
Also, tb.run() must be called.

commit: cc58cdd730b7e680ded447ee8f8cff305c83ecbb
date 3.7.2016 (dd.mm.yyyy)
OS: Ubuntu 16.04 running in a VM

github gnuradio master does not report any changes for the last 6 months for the corresponding files.

mbr0wn, 2016-09-16:
Author: Johnathan Corgan
Date: Fri, 16 September 2016 15:17:21

jmcorgan, 2016-09-17:
The original comment from redmine didn't carry over here, but this was determined to a usage error, and that it works correctly with the proper function arguments.  Closing.

closed 2016-09-17 by jmcorgan