gnuradio/gnuradio #938: Implement variable number of inputs and outputs for hierarchical blocks.

mbr0wn, 2016-09-16:
It should be possible to declare a hierarchical block to have a variable number of inputs or outputs, similar to how gr_block's behave.

For gr_block derived blocks, a C++ function, 'check_topology' is called during flowgraph initialization to allow the block to learn how many inputs and outputs have been wired to it, validate this is correct, and and perform any needed internal connections to support these values.

The equivalent function call for hierarchical blocks needs to occur earlier in the process, during hierarchical flowgraph flattening.  Also, the function call must support calling up into Python to allow Python-defined hierarchical blocks to respond.

mbr0wn, 2016-09-16:
Author: Johnathan Corgan
Date: Sun, 19 April 2009 22:02:02

mbr0wn, 2016-09-16:
Author: Ben Reynwar
Date: Tue, 08 January 2013 16:39:31

mbr0wn, 2016-09-16:
Author: Johnathan Corgan
Date: Tue, 16 July 2013 18:41:55

jmcorgan, 2016-09-17:
This will need a bit of a rewrite in the hier_block2 and flowgraph flattening areas.  Given that the request has been open for 7 years, and I was the one that requested it, I feel safe in closing this until we revisit the container code for hier blocks in general.

closed 2016-09-17 by jmcorgan