gnuradio/gnuradio #939: gr-audio windows (source not implemented)

mbr0wn, 2016-09-16:
At the moment audio_windows_source::work  has  an '#if 0' enclosing the contents of it's work function (i.e. the block does nothing).
We either need to make a decision to implement this, or delete the block.

mbr0wn, 2016-09-16:
Author: Ben Reynwar
Date: Wed, 06 March 2013 17:22:25

mbr0wn, 2016-09-16:
Author: Balint Seeber
Date: Thu, 07 March 2013  0:39:03

mbr0wn, 2016-09-16:
Author: Johnathan Corgan
Date: Sat, 22 June 2013 13:51:14

mbr0wn, 2016-09-16:
Author: Johnathan Corgan
Date: Tue, 16 July 2013 18:55:34

mbr0wn, 2016-09-16:
Author: Johnathan Corgan
Date: Fri, 30 August 2013  6:18:46

mbr0wn, 2016-09-16:
Author: Johnathan Corgan
Date: Thu, 01 May 2014 20:24:43

jmcorgan, 2016-09-17:
@gnieboer Can you take a look at this?  It is ancient code and unclear if it was ever working.

gnieboer, 2016-09-18:
Can do.  It's been on my list but haven't quite gotten there.

gnieboer, 2016-10-12:
Pull request ready for review.

jmcorgan, 2016-11-04:
Fixed in #1051

closed 2016-11-04 by jmcorgan