gnuradio/gnuradio #940: (cosmetic) gr_lfsr's constant should be 'unsigned int' instead of 'int'

mbr0wn, 2016-09-16:
Reducing 0x80000057u to 'int' datatype inside of { } is not valid in C++11,
therefore the datatype should be 'unsigned int' instead of 'int'.

mbr0wn, 2016-09-16:
Author: Johnathan Corgan
Date: Sun, 19 May 2013 18:47:17

mbr0wn, 2016-09-16:
Author: Moritz Fischer
Date: Mon, 20 May 2013 10:37:32

mbr0wn, 2016-09-16:
Author: Johnathan Corgan
Date: Mon, 20 May 2013 13:58:12

mbr0wn, 2016-09-16:
Author: Moritz Fischer
Date: Mon, 20 May 2013 15:36:48

mbr0wn, 2016-09-16:
Author: Johnathan Corgan
Date: Sat, 22 June 2013 13:56:18

mbr0wn, 2016-09-16:
Author: Johnathan Corgan
Date: Tue, 16 July 2013 19:09:57

mbr0wn, 2016-09-16:
Author: Johnathan Corgan
Date: Fri, 30 August 2013  6:18:58

mbr0wn, 2016-09-16:
Author: Johnathan Corgan
Date: Thu, 09 January 2014 23:41:16

jmcorgan, 2016-09-17:
This was completed in #893.

closed 2016-09-17 by jmcorgan