gnuradio/gnuradio #943: top_block.unlock() deadlock when flow graph contains Python sync_block

mbr0wn, 2016-09-16:
When a flow graph contains a sync_block written in Python, calling top_block.lock() and then top_block.unlock() always freezes.

Simple debugging shows the hang is caused by a thread calling sem_wait().

The attached file contains a simple repro.

mbr0wn, 2016-09-16:
Author: Johnathan Corgan
Date: Sat, 21 September 2013 15:02:32

mbr0wn, 2016-09-16:
Author: Johnathan Corgan
Date: Thu, 26 September 2013 17:46:52

mbr0wn, 2016-09-16:
Author: Tom Rondeau
Date: Sun, 29 September 2013 18:40:42

mbr0wn, 2016-09-16:
Author: Johnathan Corgan
Date: Tue, 30 September 2014 17:24:46

mbr0wn, 2016-09-16:
Author: Jiří Pinkava
Date: Thu, 12 March 2015  8:21:31

mbr0wn, 2016-09-16:
Author: Jiří Pinkava
Date: Thu, 12 March 2015  8:23:31

mbr0wn, 2016-09-16:
Author: Kevin Reid
Date: Thu, 12 March 2015 15:47:24

mbr0wn, 2016-09-16:
Author: Jiří Pinkava
Date: Sun, 15 March 2015 14:54:09

mbr0wn, 2016-09-16:
Author: Jiří Pinkava
Date: Sun, 15 March 2015 21:47:09

jmcorgan, 2016-10-02:
I am unable to replicate this bug with current 3.7.10.1 using the unconnect_test.py from the original bug report 594 in Redmine. @pinkavaj can you retest now and see if the underlying issue has been resolved somewhere along the way?

pinkavaj, 2016-10-02:
This one should be fixed few months, see test_34a_lock_unlock, test_34b_lock_unlock in gr-blocks/python/blocks/qa_hier_block2.py

jmcorgan, 2016-10-02:
Ah, yes.  Fixed in #485.

closed 2016-10-02 by jmcorgan