gnuradio/gnuradio #944: Lock/Unlock segfaulting

mbr0wn, 2016-09-16:
When lock/unlock is used to do the disconnection/reconnection
in a python top_block, I always get segfaults after at most a couple of 
changes when unlock() is called

Here is an example of this behaviour:

sinusoid--> throtle --> (ON or OFF block chosen by a "button chooser") --> null sink

http://web.eecs.umich.edu/~anastas/docs/onoff_flat_test1.py

This is under version: 3.7.2git-79-g931a7b07 / 3.7.2git on a fedora 18 64bit machine

mbr0wn, 2016-09-16:
Author: Johnathan Corgan
Date: Tue, 08 October 2013 21:56:29

mbr0wn, 2016-09-16:
Author: Johnathan Corgan
Date: Sat, 16 November 2013 17:33:53

mbr0wn, 2016-09-16:
Author: Alexandru Csete
Date: Fri, 11 April 2014 22:00:19

mbr0wn, 2016-09-16:
Author: Johnathan Corgan
Date: Tue, 29 April 2014 15:59:36

mbr0wn, 2016-09-16:
Author: Johnathan Corgan
Date: Tue, 30 September 2014 17:44:06

mbr0wn, 2016-09-16:
Author: Jiří Pinkava
Date: Wed, 10 June 2015 17:29:21

jmcorgan, 2016-10-02:
@anastas I've tried the http://web.eecs.umich.edu/~anastas/docs/onoff_flat_test1.py again with the current 3.7.10.1 release with no issues.  Can you verify this on your end?

jmcorgan, 2016-10-06:
Closing as fixed.

closed 2016-10-06 by jmcorgan