gnuradio/gnuradio #947: gr-digital: tunnel.py tune failure if lo-offset is not defined

mbr0wn, 2016-09-16:
Originally reported by Johnathan Fox, the lo-offset variable if not defined on the commandline causes the following error:

NotImplementedError: Wrong number of arguments for overloaded function 'new_tune_request_t'.

See the following for the terminal output showing the error:

[root@cobra narrowband]# ./tunnel.py -f 146.0M -a addr="10.2.8.104"
linux; GNU C++ version 4.4.7 20120313 (Red Hat 4.4.7-4); Boost_104100; UHD_003.007.001-64-g92b0b7ab

Using Volk machine: sse4_2_64
-- Opening a USRP2/N-Series device...
-- Current recv frame size: 1472 bytes
-- Current send frame size: 1472 bytes

UHD Warning:
    The recv buffer could not be resized sufficiently.
    Target sock buff size: 50000000 bytes.
    Actual sock buff size: 5000000 bytes.
    See the transport application notes on buffer resizing.
    Please run: sudo sysctl -w net.core.rmem_max=50000000

UHD Warning:
    The recv buffer could not be resized sufficiently.
    Target sock buff size: 50000000 bytes.
    Actual sock buff size: 5000000 bytes.
    See the transport application notes on buffer resizing.
    Please run: sudo sysctl -w net.core.rmem_max=50000000
-- Detecting internal GPSDO.... Found an internal GPSDO
-- found
-- Setting references to the internal GPSDO
-- Initializing time to the internal GPSDO

No gain specified.
Setting gain to 19.000000 (from [0.000000, 38.000000])
Traceback (most recent call last):
  File "./tunnel.py", line 296, in <module>
    main()
  File "./tunnel.py", line 259, in main
    options)
  File "./tunnel.py", line 103, in __init__
    options.verbose)
  File "/home/fox/Documents/GNU_radio/Cranial/narrowband/uhd_interface.py", line 199, in __init__
    freq, lo_offset, gain, spec, antenna, clock_source)
  File "/home/fox/Documents/GNU_radio/Cranial/narrowband/uhd_interface.py", line 70, in __init__
    self._freq = self.set_freq(freq, lo_offset)
  File "/home/fox/Documents/GNU_radio/Cranial/narrowband/uhd_interface.py", line 120, in set_freq
    r = self.u.set_center_freq(uhd.tune_request(freq, lo_offset))
  File "/usr/local/lib64/python2.6/site-packages/gnuradio/uhd/__init__.py", line 52, in __init__
    super(tune_request_t, self).__init__(*args)
  File "/usr/local/lib64/python2.6/site-packages/gnuradio/uhd/uhd_swig.py", line 792, in __init__
    this = _uhd_swig.new_tune_request_t(*args)
NotImplementedError: Wrong number of arguments for overloaded function 'new_tune_request_t'.
  Possible C/C++ prototypes are:
    uhd::tune_request_t(double)
    uhd::tune_request_t(double,double)

mbr0wn, 2016-09-16:
Author: Johnathan Corgan
Date: Tue, 30 September 2014 18:19:27

mbr0wn, 2016-09-16:
Author: Mike Jameson
Date: Mon, 05 January 2015 16:27:53

devnulling, 2020-01-21:
Closing this as tunnel.py has been removed.

closed 2020-01-21 by devnulling