gnuradio/gnuradio #949: Use of freed buffers in vmcircbuf_prefs causes invalid file names to be used

mbr0wn, 2016-09-16:
When running gnuradio 3.7.2.2 on Windows, I often get messages such as these on the console:

J??8?*\supersat\AppData\Roaming\.gnuradio\prefs\vmcircbuf_default_factory: Invalid argument
x?*?²??\supersat\AppData\Roaming\.gnuradio\prefs\vmcircbuf_default_factory: Invalid argument

Sometimes this causes random files to be dropped in the current directory.

Looking at vmcircbuf_prefs.cc, it's pretty clear what the problem is. There's a couple of freed buffers being passed around:

1. pathname() returns a C string pointer from a C++ string that goes out of scope when the function returns.
2. vmcircbuf_prefs::get returns a pointer to a buffer that's allocated on the stack, which goes out of scope when the function returns.

I would fix this but I have not (yet) built GNU Radio on Windows.

mbr0wn, 2016-09-16:
Author: Jiří Pinkava
Date: Thu, 12 March 2015 10:05:20

jmcorgan, 2016-09-18:
This was fixed in #820.

closed 2016-09-18 by jmcorgan