gnuradio/gnuradio #951: Handle buffers which are not doubly-mapped appropriately

mbr0wn, 2016-09-16:
In the case that a block requires a custom buffer allocator (#730), but is unable to provide a doubly-mapped buffer, this should be handled correctly by the scheduler: e.g. by ensuring that @work()@ is never called with a number of items that would cause the edge of the buffer to be crossed.

Based on discussions held during Co-Processor Working Group session(s) ([[GRCon14CoProc]], [[GRCon13Coprocessor]])

mbr0wn, 2016-09-16:
Author: Michael Dickens
Date: Wed, 24 September 2014 12:26:09

mbr0wn, 2016-09-16:
Author: Michael Dickens
Date: Wed, 26 August 2015 19:45:59

mbr0wn, 2016-09-17:
@douggeiger

jmcorgan, 2016-09-17:
This work was taken over by @sdh11 and is a work in progress.  Currently, is being tracked by PR #741

mbr0wn, 2021-05-14:
This is being handled here: https://github.com/gnuradio/gnuradio-ngsched

closed 2021-05-14 by mbr0wn