gnuradio/gnuradio #953: Make file_source take an offset and length to read

mbr0wn, 2016-09-16:
I'd like to be able to peek at small intervals from large recordings. I propose adding two parameters to file_source: offset and readlen

Offset n = 0 would preserve the current behavior; offset n > 0 would seek n items into the file (n*1 byte for chars, n*2 bytes for shorts, n*4 bytes for floats)
Readlen = 0 would preserve the current behavior; readlen n > 0 would effectively return a substring from the file (1 = give me 1 item, 1e6 = give me the next million items), then do whatever happens when EOF is reached; repeat or end.

I'd rather do this with a source block than by slicing up my source file with dd so that i can change the part of the file I'm reading at runtime, and so that I don't accidentally split and interesting signal in the middle.

mbr0wn, 2016-09-16:
Author: Nathan West
Date: Fri, 03 October 2014  8:59:35

mbr0wn, 2016-09-16:
Author: Chris Kuethe
Date: Fri, 03 October 2014 18:54:55

mbr0wn, 2016-09-16:
Author: Jeff Long
Date: Sat, 04 October 2014 16:32:58

mbr0wn, 2016-09-16:
Author: Chris Kuethe
Date: Sat, 04 October 2014 18:18:58

mbr0wn, 2016-09-16:
Author: Chris Kuethe
Date: Sat, 04 October 2014 22:24:31

mbr0wn, 2016-09-16:
Author: Kevin Reid
Date: Sun, 05 October 2014 14:26:16

mbr0wn, 2016-09-16:
Author: Jeff Long
Date: Sun, 05 October 2014 15:40:41

mbr0wn, 2016-09-16:
Author: Chris Kuethe
Date: Sun, 05 October 2014 16:45:39

mbr0wn, 2016-09-16:
Author: Chris Kuethe
Date: Sun, 05 October 2014 17:04:10

mbr0wn, 2016-09-16:
Author: Jeff Long
Date: Sun, 05 October 2014 17:35:42

mbr0wn, 2016-09-16:
Author: Chris Kuethe
Date: Sun, 05 October 2014 19:32:42

mbr0wn, 2016-09-16:
Author: Jeff Long
Date: Sun, 05 October 2014 21:47:51

mbr0wn, 2016-09-16:
Author: Chris Kuethe
Date: Mon, 06 October 2014  5:54:23

mbr0wn, 2016-09-16:
Author: Johnathan Corgan
Date: Mon, 13 October 2014 19:29:05

noc0lour, 2017-11-02:
Is there still interest in this feature? We could tag this as easy I'd say?

mbr0wn, 2017-11-10:
Good idea. Would have been good for hacktoberfest...

willcode, 2018-01-16:
Submitted https://github.com/gnuradio/gnuradio/pull/1550 against next.

marcusmueller, 2018-02-11:
Closing since #1550 was manually merged.

closed 2018-02-11 by marcusmueller