gnuradio/gnuradio #955: Resources in .pyd files

mbr0wn, 2016-09-16:
It would be nice to have a resource section in the _Python.pyd files. This could be a hint to find .pyd files with 
a mismatching version. (I'm not sure if some Python/Swig file could identify such a version mis-match).

I've done this for myself on Windows using the already generated lib/gnuradio*.rc files. 
Since CMake is a black art to me, I've created some swig/makefile.
E.g. for MSVC in gr-blocks/swig/Makefile:

<pre>
$(OBJ_DIR)/blocks_swig%.res: ../lib/gnuradio-blocks.rc
      sed -e 's/gnuradio-blocks/_blocks_swig$*/' \
          -e 's/.dll/.pyd/'           < $< > gnuradio-tmp.rc
      rc $(RCFLAGS) -fo $@ gnuradio-tmp.rc
</pre>

And use those .res files in the link step.

The output from e.g. filever (similar would show up in Explorer):
<pre>
filever -v _blocks_swig0.pyd
--a-- W32i   DLL ENU         3.7.7.6 shp  1,812,992 10-17-2014 _blocks_swig0.pyd
        Language        0x0409 (Engelsk (USA))
        CharSet         0x04b0 Unicode
        OleSelfRegister Disabled
        FileDescription _blocks_swig0
        InternalName    _blocks_swig0.pyd
        OriginalFilenam _blocks_swig0.pyd
        ProductName     _blocks_swig0
        ProductVersion  3.7.6
        FileVersion     3.7.6
        LegalCopyright  Licensed under GPLv3 or any later version

        VS_FIXEDFILEINFO:
        Signature:      feef04bd
        Struc Ver:      00010000
        FileVer:        00030007:00070006 (3.7:7.6)
        ProdVer:        00030007:00070006 (3.7:7.6)
        FlagMask:       0000003f
        Flags:          00000000
        OS:             00000004 Win32
        FileType:       00000002 Dll
        SubType:        00000008
        FileDate:       00000000:00000000
</pre>

All this assumes off-course that ../lib/gnuradio-blocks.dll has already been linked.
Maybe the "FileDescription" should contain some more sensible description. Like:
"Python module _blocks_swig0".

mbr0wn, 2016-09-16:
Author: Martin Braun
Date: Sat, 25 October 2014 12:30:51

marcusmueller, 2020-10-29:
Closing due to move away from SWIG requiring reformulation.

closed 2020-10-29 by marcusmueller