gnuradio/gnuradio #957: WX-based apps will die complaining of C++ assertion m_window failed in GetSize()

mbr0wn, 2016-09-16:
It was suggested that I either update an existing bug or submit this as a bug. I have not found reference to "m_window" or GetSize() in the buglist here.

There is a bug in GR with regard to wxPython 3.0 - it will winge about "m_window" and fail in any wxgui.

This only appears in "Debug" build: in "Release" build the application will fail silently.

wx._core.PyAssertionError: C++ assertion "m_window" failed at
./src/gtk/dcclient.cpp(2041) in DoGetSize(): GetSize() doesn't work
without window

There is a workaround for any given application that was here: http://imyy.in/blog/?p=26
However, this particular link is now gone. 

The workaround showed how to fix any particular application. I have made this more general by putting it the wxgui top_block::__init__ itself, which looks to fix GR WX widgets without breaking WX in general. The removed "Start()" method looks to be called again after the window is instanced.

$ diff -btwc  ../gr-wxgui/grc/top_block_gui.py.orig ../gr-wxgui/grc/top_block_gui.py
*** ../gr-wxgui/grc/top_block_gui.py.orig       Mon Jul  7 17:28:59 2014
--- ../gr-wxgui/grc/top_block_gui.py    Sat Nov  8 13:02:32 2014
***************
*** 47,52 ****
--- 47,53 ----
                  self.Add = self._panel.Add
                  self.GridAdd = self._panel.GridAdd
                  self.GetWin = self._panel.GetWin
+                 self.Start(False)
 
          def SetIcon(self, *args, **kwargs): self._frame.SetIcon(*args, **kwargs)

My baseline and requirements.

* This is GnuRadio 3.7.5.1 on CentOS 7.0
* need CentOS/RHEL baseline for a variety of reasons
* really rather have WX as other code uses it
* pybombs did not seem to do the job
* built from gnuradio git sources
* UHD 3.7.1
* osmocore and rtl-sdr gr-osmosdr form osmocom.org
* needs qwt from EPEL repository
* needs PyQwt from sourceforge

 
----------- Additional correspondence from the discuss-gnuradio forum:

Can you add this info to the GNU Radio Issue tracker? Look to see if there's an issue open about the WX 3 problem, and if not, add one with what you have here. I can easily test this on WX 2.8, and if it fails, we can try to put in proper checks to do the "right thing" for each version.

Thanks,
Tom

On Mon, Feb 9, 2015 at 9:56 AM, Red Gator <redgatormds@gmail.com> wrote:
Sorry it's been so long to respond... been mostly "out"...

I have tested this with WX 3.0 and the GR apps. It appears to be a mismatch between the way GR instances and initializes WX 3.0 objects and the way WX expects to be instanced. The ::Start operation is actually done again later, so there is no side effect except exception. Note the exception only shows up in Debug mode... in Release mode GR quietly dies with no diagnostic messages at all (which is IMHO a very annoying behavior). I haven't tried this with WX <3.0.

On Mon, Nov 10, 2014 at 11:05 AM, Tom Rondeau <tom@trondeau.com> wrote:
On Sat, Nov 8, 2014 at 4:39 PM, Red Gator <redgatormds@gmail.com> wrote:
Just got this limping around.

* need CentOS/RHEL baseline for a variety of reasons
* really rather have WX as other code uses it
* pybombs did not seem to do the job
* built from gnuradio git sources
* UHD 3.7.1
* osmocore and rtl-sdr gr-osmosdr form osmocom.org
* needs qwt from EPEL repository
* needs PyQwt from sourceforge
* There is a bug in GR with regard to wxPython 3.0 - it will winge about "m_window" and fail in any wxgui.

wx._core.PyAssertionError: C++ assertion "m_window" failed at
./src/gtk/dcclient.cpp(2041) in DoGetSize(): GetSize() doesn't work
without window
Yes, we've known about this but not one has submitted a patch to us to fix it.
 
There is a workaround for any given application here: http://imyy.in/blog/?p=26  

It seems that this can be made more general by putting it the wxgui top_block::__init__ itself:

$ diff -btwc  ../gr-wxgui/grc/top_block_gui.py.orig ../gr-wxgui/grc/top_block_gui.py
*** ../gr-wxgui/grc/top_block_gui.py.orig       Mon Jul  7 17:28:59 2014
--- ../gr-wxgui/grc/top_block_gui.py    Sat Nov  8 13:02:32 2014
***************
*** 47,52 ****
--- 47,53 ----
                  self.Add = self._panel.Add
                  self.GridAdd = self._panel.GridAdd
                  self.GetWin = self._panel.GetWin
+                 self.Start(False)
 
          def SetIcon(self, *args, **kwargs): self._frame.SetIcon(*args, **kwargs)
 

Ok, that's great that there seems to be a fix out there for the WX 3.0 problem. Will this cause any other side effects in the applications? Will it work for WX < 3.0?

Tom

mbr0wn, 2016-09-16:
Author: Martin Braun
Date: Fri, 13 November 2015 18:59:01

jmcorgan, 2016-09-17:
Since wxgui has reached end of life and has been removed for 3.8, I am closing this.

closed 2016-09-17 by jmcorgan