gnuradio/gnuradio #958: Rename finished() member function in block.h to is_msg_handler_finished() or something similar

mbr0wn, 2016-09-16:
To reduce confusing, it is desirable to rename the finished() function to something more descriptive. #730 requires the addition of several life-cycle management functions for blocks, and the current proposed named for the function that will handle deallocating memory is finish(). This is potentially confusing, and since the the finished() function is used to indicate that the message port handler is complete, a rename is suggested.
This is an API breaking change, so any rename will have to wait until release 3.8.0

mbr0wn, 2016-09-16:
Author: Martin Braun
Date: Tue, 24 November 2015 22:45:06

marcusmueller, 2023-10-24:
Not in scope for 3.x anymore, I'm afraid

closed 2023-10-24 by marcusmueller