gnuradio/gnuradio #961: Segfault using gr-mac and pmts. Race condition?

mbr0wn, 2016-09-16:
Running the attached script with GNU Radio from maint, segfaults after some time. I did not attach the core as it's 76M in size.
This was run on 1c33a22e92654bb8333fcd0360414a514e6927f8 (maint) on a oe dizzy based Xilinx Zynq system.

Steps to reproduce:

1) Build off of maint / or master
2) python pmt_smasher.py
3) Wait ... segfault

> This GDB was configured as "arm-oe-linux-gnueabi".
> Type "show configuration" for configuration details.
> For bug reporting instructions, please see:
> <http://www.gnu.org/software/gdb/bugs/>.
> Find the GDB manual and other documentation resources online at:
> <http://www.gnu.org/software/gdb/documentation/>.
> For help, type "help".
> Type "apropos word" to search for commands related to "word"...
> Reading symbols from python...Reading symbols from /usr/bin/.debug/python2.7...done.
> done.
> (gdb) set args pmt_smasher.py
> (gdb) core-file core
> warning: core file may not match specified executable file.
> [New LWP 1014]
> [New LWP 1011]
> [New LWP 1009]
> [New LWP 1013]
> [New LWP 1012]
> [New LWP 1010]
> [Thread debugging using libthread_db enabled]
> Using host libthread_db library "/lib/libthread_db.so.1".
> Core was generated by `python ./pmt_smasher.py'.
> Program terminated with signal SIGSEGV, Segmentation fault.
> #0  0x0000001c in ?? ()
> (gdb) bt
> #0  0x0000001c in ?? ()
> #1  0xb5438768 in ~intrusive_ptr (this=0xb2900748, __in_chrg=<optimized out>)  
> at /home/mfischer/local/oecore-x86_64-pmt-dizzy/sysroots/armv7ahf-vfp-neon-oe-linux-gnueabi/usr/include/boost/smart_ptr/intrusive_ptr.hpp:97
> #2  _wrap_delete_swig_int_ptr (args=<optimized out>)
> at /home/mfischer/src/gnuradio.git/build-pmt-dizzy/gnuradio-runtime/swig/pmt_swigPYTHON_wrap.cxx:30147
> #3  0xb6e55a60 in PyObject_Call (func=func@entry=0xd41e18,
> arg=arg@entry=0x14bee70, kw=kw@entry=0x0) at Objects/abstract.c:2529
> #4  0xb6e5602c in PyObject_CallFunctionObjArgs (callable=0xd41e18)
> at Objects/abstract.c:2760
> #5  0xb542f9b4 in SwigPyObject_dealloc (v=0x1b2c0b0)
> at /home/mfischer/src/gnuradio.git/build-pmt-dizzy/gnuradio-runtime/swig/pmt_swigPYTHON_wrap.cxx:1676
> #6  0xb6e93130 in dict_dealloc (mp=0x1b2bed0) at Objects/dictobject.c:985
> #7  0xb6eb0a20 in subtype_dealloc (self=0x1b25e70) at Objects/typeobject.c:999
> #8  0xb6ef9710 in call_function (oparg=<optimized out>, pp_stack=0xb17fe700)
> at Python/ceval.c:4055
> #9  PyEval_EvalFrameEx (f=f@entry=0xb2900bd8, throwflag=throwflag@entry=0)
> at Python/ceval.c:2666
> #10 0xb6efbde8 in fast_function (nk=<optimized out>, na=<optimized out>, n=2,
> pp_stack=0xb17fe7c8, func=<optimized out>) at Python/ceval.c:4107
> #11 call_function (oparg=<optimized out>, pp_stack=0xb17fe7c8)
> ---Type <return> to continue, or q <return> to quit---
> at Python/ceval.c:4042
> #12 PyEval_EvalFrameEx (f=f@entry=0x0, throwflag=<optimized out>)
> at Python/ceval.c:2666
> #13 0xb6efc9b8 in PyEval_EvalCodeEx (co=0x108b2a8, globals=<optimized out>,
> locals=locals@entry=0x0, args=args@entry=0x1b2a384, argcount=2,
> kws=kws@entry=0x0, kwcount=kwcount@entry=0, defs=defs@entry=0x0,
> defcount=defcount@entry=0, closure=closure@entry=0x0)
> at Python/ceval.c:3253
> #14 0xb6e7fea8 in function_call (func=0x108edf0, arg=0x1b2a378, kw=0x0)
> at Objects/funcobject.c:526
> #15 0xb6e55a60 in PyObject_Call (func=func@entry=0x108edf0,
> arg=arg@entry=0x1b2a378, kw=kw@entry=0x0) at Objects/abstract.c:2529
> #16 0xb6e682d8 in instancemethod_call (func=0x108edf0, arg=0x1b2a378, kw=0x0)
> at Objects/classobject.c:2578
> #17 0xb6e55a60 in PyObject_Call (func=func@entry=0x18c6968,
> arg=arg@entry=0x14beef0, kw=kw@entry=0x0) at Objects/abstract.c:2529
> 
> #18 0xb6e55f7c in PyObject_CallMethodObjArgs (callable=0x18c6968,
> name=0x1b25dc0) at Objects/abstract.c:2738
> #19 0xb55bc884 in SwigDirector_feval_p::eval (this=0x1b5f728, x=...)
> at /home/mfischer/src/gnuradio.git/build-pmt-dizzy/gnuradio-runtime/swig/runtime_swigPYTHON_wrap.cxx:7074
> #20 0xb55ef490 in gr::py_feval_p::calleval (this=0x1b5f728, x=...)
> at /home/mfischer/src/gnuradio.git/gnuradio-runtime/include/gnuradio/py_feval.h:88
> #21 0xb677b398 in gr::block_gateway::dispatch_msg (this=0x1b5e7c8,
> which_port=..., msg=...)
> at /home/mfischer/src/gnuradio.git/gnuradio-runtime/include/gnuradio/block_gateway.h:294
> #22 0xb67b7374 in gr::tpb_thread_body::tpb_thread_body (this=0xb17fecf8,
> block=..., max_noutput_items=<optimized out>)
> at /home/mfischer/src/gnuradio.git/gnuradio-runtime/lib/tpb_thread_body.cc:106
> #23 0xb67adb98 in operator() (this=0x1948460)
> at /home/mfischer/src/gnuradio.git/gnuradio-runtime/lib/scheduler_tpb.cc:44
> #24 gr::thread::thread_body_wrapper<gr::tpb_container>::operator() (
> this=0x1948460)
> at /home/mfischer/src/gnuradio.git/gnuradio-runtime/include/gnuradio/thread/thread_body_wrapper.h:51
> #25 0xb6760970 in operator() (this=<optimized out>)
> at /home/mfischer/local/oecore-x86_64-pmt-dizzy/sysroots/armv7ahf-vfp-neon-oe-linux-gnueabi/usr/include/boost/function/function_template.hpp:767
> #26 boost::detail::thread_data<boost::function0<void> >::run (
> this=<optimized out>)
> at /home/mfischer/local/oecore-x86_64-pmt-dizzy/sysroots/armv7ahf-vfp-neon-oe-linux-gnueabi/usr/include/boost/thread/detail/thread.hpp:115
> #27 0xb655bf68 in boost::(anonymous namespace)::thread_proxy (
> #28 0xb6cd0db0 in start_thread (arg=0xb17ff460) at pthread_create.c:315
> #29 0xb6db9080 in ?? () at ../sysdeps/unix/sysv/linux/arm/clone.S:89
> from /lib/libc.so.6

mbr0wn, 2016-09-16:
Author: Marcus Müller
Date: Sat, 21 March 2015 15:31:25

mbr0wn, 2016-09-16:
Author: Marcus Müller
Date: Sun, 22 March 2015  7:43:45

mbr0wn, 2016-09-16:
Author: Marcus Müller
Date: Sun, 22 March 2015 10:05:13

mbr0wn, 2016-09-16:
Author: Doug Geiger
Date: Sun, 22 March 2015 18:23:19

mbr0wn, 2016-09-16:
Author: Marcus Müller
Date: Sun, 22 March 2015 18:51:09

mbr0wn, 2016-09-16:
Author: Moritz Fischer
Date: Tue, 31 March 2015  0:06:11

mbr0wn, 2016-09-16:
Author: Moritz Fischer
Date: Tue, 31 March 2015  0:07:46

mbr0wn, 2016-09-16:
Author: Moritz Fischer
Date: Tue, 31 March 2015  0:39:32

mbr0wn, 2016-09-16:
Author: Marcus Müller
Date: Tue, 31 March 2015 13:10:46

darek-kawamoto, 2017-02-22:
Oh hey I think I fixed this one.

marcusmueller, 2020-10-29:
> Oh hey I think I fixed this one.

In that case, let us close that. Also, yeah, I'm not aware of segfaulting unit tests in that domain.

closed 2020-10-29 by marcusmueller