gnuradio/gnuradio #963: Trigger delay causes QT TimeSink to use excessive CPU with tags and high sample rate

mbr0wn, 2016-09-16:
The use of trigger delay causes the QT TimeSink to consume excessive CPU in the case of a high sample rate and periodic tagged bursts.

See the attached GRC, ps output, and OProfile opreport output.

It appears that the tag manipulations performed by time_sink_f_impl::_reset() in the case of a delayed trigger, as currently implemented, can be quite expensive, as they cause a significant amount of time to be spent in pmt::intrusive_ptr_add_ref(pmt::pmt_base*) and pmt::intrusive_ptr_release(pmt::pmt_base*).

mbr0wn, 2016-09-16:
Author: Tom Rondeau
Date: Thu, 09 April 2015 15:05:17

mbr0wn, 2016-09-16:
Author: Tom Rondeau
Date: Thu, 09 April 2015 15:05:27

marcusmueller, 2020-10-30:
This has gotten better through slightly improved internal taghandling. 

The way we handle tags in `time_sink_f` is still not great. Enabling *trigger delay* does make the rate you can push through a qt time sink drop by around a factor of 10 (tags every 103 items).