gnuradio/gnuradio #964: CMake warnings with CMake 3.0.2

mbr0wn, 2016-09-16:
CMake 3.0.2 (as shipped with XUbuntu 15.04) reports the following CMake policy warnings during configuration of a GNU Radio 3.7.7 build.

"CMP0026":http://www.cmake.org/cmake/help/v3.0/policy/CMP0026.html#cmp0026 - @Disallow use of the LOCATION target property.@
"CMP0043":http://www.cmake.org/cmake/help/v3.0/policy/CMP0043.html#cmp0043 - @Ignore COMPILE_DEFINITIONS_<Config> properties@
"CMP0045":http://www.cmake.org/cmake/help/v3.0/policy/CMP0045.html#cmp0045 - @Error on non-existent target in get_target_property.@
"CMP0046":http://www.cmake.org/cmake/help/v3.0/policy/CMP0046.html#cmp0046 - @Error on non-existent dependency in add_dependencies.@

I seemed to be still be running into these when setting @cmake_policy(VERSION 2.8.12)@ in the top-level @CMakeLists.txt@, but still need to dig further.

The full log is shown below.

<pre>
-- The CXX compiler identification is GNU 4.9.2
-- The C compiler identification is GNU 4.9.2
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Build type not specified: defaulting to release.
-- Build type set to Release.
-- Found Git: /usr/bin/git  
-- Extracting version information from git describe...
-- Performing Test HAVE_VISIBILITY_HIDDEN
-- Performing Test HAVE_VISIBILITY_HIDDEN - Success
-- Performing Test HAVE_WARN_SIGN_COMPARE
-- Performing Test HAVE_WARN_SIGN_COMPARE - Success
-- Performing Test HAVE_WARN_ALL
-- Performing Test HAVE_WARN_ALL - Success
-- Performing Test HAVE_WARN_NO_UNINITIALIZED
-- Performing Test HAVE_WARN_NO_UNINITIALIZED - Success
-- Compiler Version: cc (Ubuntu 4.9.2-10ubuntu13) 4.9.2
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
-- Compiler Flags: /usr/bin/cc:::-O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
/usr/bin/c++:::-O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
-- ADDING PERF COUNTERS
-- Building Static Libraries: OFF
-- Boost version: 1.55.0
-- Found the following Boost libraries:
--   date_time
--   program_options
--   filesystem
--   system
--   thread
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython2.7.so (found suitable version "2.7.9", minimum required is "2") 
-- 
-- Checking for module SWIG
-- Found SWIG version 2.0.12.
-- Found SWIG: /usr/bin/swig2.0  
-- 
-- The build system will automatically enable all components.
-- Use -DENABLE_DEFAULT=OFF to disable components by default.
-- 
-- Configuring python-support support...
--   Dependency PYTHONLIBS_FOUND = TRUE
--   Dependency SWIG_FOUND = TRUE
--   Dependency SWIG_VERSION_CHECK = TRUE
--   Enabling python-support support.
--   Override with -DENABLE_PYTHON=ON/OFF
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.28") 
-- checking for module 'cppunit'
--   found cppunit, version 1.13.2
-- Found CPPUNIT: /usr/lib/x86_64-linux-gnu/libcppunit.so;dl  
-- 
-- Configuring testing-support support...
[... 1173 lines cut ...]
  get_target_property() called with non-existent target "/usr/bin/python2".
Call Stack (most recent call first):
  gr-blocks/python/blocks/CMakeLists.txt:57 (GR_ADD_TEST)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at cmake/Modules/GrTest.cmake:45 (get_target_property):
  Policy CMP0045 is not set: Error on non-existent target in
  get_target_property.  Run "cmake --help-policy CMP0045" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  get_target_property() called with non-existent target "-B".
Call Stack (most recent call first):
  gr-blocks/python/blocks/CMakeLists.txt:57 (GR_ADD_TEST)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at cmake/Modules/GrTest.cmake:45 (get_target_property):
  Policy CMP0045 is not set: Error on non-existent target in
  get_target_property.  Run "cmake --help-policy CMP0045" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  get_target_property() called with non-existent target
  "/home/jon/software/gnuradio/gnuradio/gr-blocks/python/blocks/qa_transcendental.py".
Call Stack (most recent call first):
  gr-blocks/python/blocks/CMakeLists.txt:57 (GR_ADD_TEST)
This warning is for project developers.  Use -Wno-dev to suppress

mbr0wn, 2016-09-16:
Author: Jon Szymaniak
Date: Thu, 30 April 2015 22:18:44

mbr0wn, 2016-09-16:
Author: Michael Dickens
Date: Fri, 01 May 2015  1:46:44

mbr0wn, 2016-09-17:
@michaelld

michaelld, 2016-09-17:
@mb0wn: any other history on this one? those cmake warnings are normal & can be suppressed pretty easily. it'll take some hacking to get rid of them. cmake changes policies with respect to get_target_property a long time ago. just haven't had enough time to get it fixed.

jmcorgan, 2016-09-17:
I think this was already fixed.

michaelld, 2016-09-17:
@jmcorgan could be that too. seems pretty old. I know that the get_target_property for python scripts still produces these warnings. sooner or later cmake will deprecate GR's use of this function ... better to get it fixed sooner rather than later. I'll add this to my queue ... which is already too long ....

jmcorgan, 2016-09-17:
I'm on my phone but look for a commit by Stefan Wunsch that conditionalizes the policy check.

michaelld, 2016-09-17:
Will do. I'll admit that I haven't checked this issue inside GR for a while. I still see it in OOTs. No great surprise there ...

michaelld, 2016-10-06:
Yes, looks like those warnings were fixed in the GR repo. They still exist in OOT modules, though those of course might use old cmake code & the best we can do is make sure gr_modtool uses the new cmake codes (which I think it does).

Anyway, I think this ticket can be closed.

jmcorgan, 2016-10-06:
This was fixed in #840.

closed 2016-10-06 by jmcorgan