gnuradio/gnuradio #966: Message Blocks without data never exit from a limited data source

mbr0wn, 2016-09-16:
[Problem]
In GR 3.7.7.1, blocks which process async messages do not exit properly when given sources with limited data.

[Examples]
Example 1: pdu_to_tagged_stream
Does not exit properly when using a source with limited data. See attached example courtesy Bastian Bloessl.

Example 2: wireshark_connector in gr-foo
Example GRC file, requires gr-80211 OOTM (https://github.com/bastibl/gr-ieee802-11)
https://github.com/garverp/gr-ieee802-11/blob/offline_processing/examples/playback_wifi_rx_cli.grc

[Investigation]
See mailing list discussion: https://lists.gnu.org/archive/html/discuss-gnuradio/2015-06/msg00060.html

Both example 1 and 2 exhibit the same problem. While the "done" message is received by system_handler, the work() function continues to be called. Since this occurs with two different blocks, I propose this is not an application-specific issue, which has not been solved by https://github.com/gnuradio/gnuradio/commit/035b9d016dffefec890323bd0b24dbaf23aa9186.

mbr0wn, 2016-09-16:
Author: Paul Garver
Date: Mon, 17 August 2015 19:55:58

bastibl, 2016-09-17:
This was supposed to be addressed by this pull request

https://github.com/gnuradio/gnuradio/pull/797

mbr0wn, 2016-09-17:
It's possible this was fixed but not closed. We just auto-imported all the issues from Redmine. If it's fixed, we can close np. @bastibl can you comment if that's true?

jmcorgan, 2016-09-17:
It's fixed.

closed 2016-09-17 by jmcorgan