gnuradio/gnuradio #967: Visual Studio compile errors ('""' is not recognized as an internal or external command)

mbr0wn, 2016-09-16:
Apologies for the creation of #798, I can't find a way to delete it.

When compiling the gnu-radio project in Visual studio (using CMake 3.2.2 and Visual-Studio 12.0.31101.00 Update 4), many projects fail to compile due to custom build script errors. (I've found two other reports of this in the discuss-gnuradio mailing list).

The problem seems to be a custom build step against many of the .rule files that just consists of a pair of double quotes, e.g.:

 setlocal
 ""
 if %errorlevel% neq 0 goto :cmEnd
 :cmEnd
 endlocal & call :cmErrorLevel %errorlevel% & goto :cmDone
 :cmErrorLevel
 exit /b %1
 :cmDone
 if %errorlevel% neq 0 goto :VCEnd

I think these custom build rules are generated by the following stanza in \cmake\Modules\UseSWIG.cmake

  foreach(swig_gen_file ${${outfiles}})
    add_custom_command(
      OUTPUT ${swig_gen_file}
      COMMAND ""
      DEPENDS ${_target}
      COMMENT ""
    )
  endforeach()

It seems that if you put something in the COMMAND "" quotes it is passed into the custom build without the quotes, but if you just put a pair of empty double quotes as above, you get the double quotes in the custom-build script and Visual Studio throws up the compile error.
Removing or commenting-out the COMMAND "" line seems to solve the problem.

mbr0wn, 2016-09-16:
Author: Tom Rondeau
Date: Wed, 29 July 2015 18:43:32

mbr0wn, 2016-09-16:
Author: Craig Easton
Date: Thu, 30 July 2015 14:50:34

mbr0wn, 2016-09-16:
Author: Tom Rondeau
Date: Thu, 30 July 2015 14:56:12

mbr0wn, 2016-09-16:
Author: Craig Easton
Date: Mon, 03 August 2015 19:17:52

marcusmueller, 2020-10-30:
Must have been fixed by now, or obsoleted by obsoleting older MSVC versions.

closed 2020-10-30 by marcusmueller