gnuradio/gnuradio #970: runtime: a lot of things log to cout instead of using gr::logger

mbr0wn, 2016-09-16:
Extending on #806 ; this shouldn't be the case.

This will require:

* adding @#include <gnuradio/logger>@ to a lot of .cc 
* potentially adding quite some d_logger members (e.g. to @gr::flowgraph@)

kartikp1995, 2016-12-31:
Hi,
I saw lots of `cout` in around 5 files in `gnuradio-runtime`. But in each case, they have the condition before printing. That condition is based on the variable called `FILENAME_DEBUG`. So, is the assumption of all these to be DEBUG messages is valid?
Example: [https://github.com/gnuradio/gnuradio/blob/master/gnuradio-runtime/lib/hier_block2_detail.cc#L91](https://github.com/gnuradio/gnuradio/blob/master/gnuradio-runtime/lib/hier_block2_detail.cc#L91)

On the other hand, in `gr-blocks` have `cout`s mostly with variable that decides whether to display the message on the console or not. In that case they are simply information messages. Is that correct?
Example: [https://github.com/gnuradio/gnuradio/blob/master/gr-blocks/lib/tag_debug_impl.cc#L104-L107](https://github.com/gnuradio/gnuradio/blob/master/gr-blocks/lib/tag_debug_impl.cc#L104-L107)

At last, how is this the extension of #806 ?

Thanks for the clarifications.

jmcorgan, 2016-12-31:
Thanks for looking at this.

Yes, the cout's in gnuradio-runtime in several places were designed to be compiled out using ifdef's, and unless there is some exception can be treated as DEBUG level logger output.  However, IIRC, the use of the newer gr::logger facility inside the runtime itself might be problematic due to needing to manage a logger instance inside the code. It's been a long while since I've looked at this.

Other areas, as you've noticed, are using stdout/stderr in ways that could be easily substituted with gr::logger, and should by default be INFO. However, each case should be reevaluated on its own to decide what the appropriate log level should be.

The reference to 806 is the old tracking system and doesn't track the new github issues, so just ignore.

Are you considering tackling some of these?

kartikp1995, 2016-12-31:
Hi,
I can look into it. I will start with `gnuradio-runtime`. I will ask for any help if needed.

Thanks.

marcusmueller, 2020-10-30:
As of now, we have no active-unless-explicitly-debugging-something-ancient-via-`-DDEBUG_HIERBLOCK2`-or-so cout printers. Yay!

closed 2020-10-30 by marcusmueller