gnuradio/gnuradio #971: gr-digital: qa_ofdm_sync_sc_cfb sometimes passes, sometimes fails

mbr0wn, 2016-09-16:
commit 9e133ccb8fd4fadfe5cda68ae7240b088dce5230 (master of 2015-07-12)

Fedora 22, x86_64, Boost 1.57.0

<pre>
#10020|marcus>ctest -vv -R ofdm_sync                                                                                                                                [10020:8:0ms:/home/marcus/src/gnuradio/build]
Test project /home/marcus/src/gnuradio/build
    Start 151: qa_ofdm_sync_sc_cfb
1/1 Test #151: qa_ofdm_sync_sc_cfb ..............   Passed    0.35 sec

100% tests passed, 0 tests failed out of 1

Total Test time (real) =   0.37 sec
#10020|marcus>ctest -vv -R ofdm_sync                                                                                                                                [10020:0:0ms:/home/marcus/src/gnuradio/build]
Test project /home/marcus/src/gnuradio/build
    Start 151: qa_ofdm_sync_sc_cfb
1/1 Test #151: qa_ofdm_sync_sc_cfb ..............***Failed    0.32 sec

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   0.33 sec

The following tests FAILED:
	151 - qa_ofdm_sync_sc_cfb (Failed)
Errors while running CTest
#10020|marcus>ctest -vv -R ofdm_sync                                                                                                                                [10020:8:0ms:/home/marcus/src/gnuradio/build]
Test project /home/marcus/src/gnuradio/build
    Start 151: qa_ofdm_sync_sc_cfb
1/1 Test #151: qa_ofdm_sync_sc_cfb ..............   Passed    0.33 sec

100% tests passed, 0 tests failed out of 1

Total Test time (real) =   0.34 sec
#10020|marcus>ctest -vv -R ofdm_sync                                                                                                                                [10020:0:0ms:/home/marcus/src/gnuradio/build]
Test project /home/marcus/src/gnuradio/build
    Start 151: qa_ofdm_sync_sc_cfb
1/1 Test #151: qa_ofdm_sync_sc_cfb ..............***Failed    0.33 sec

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   0.34 sec

The following tests FAILED:
	151 - qa_ofdm_sync_sc_cfb (Failed)
Errors while running CTest
#10020|marcus>ctest -vv -R ofdm_sync                                                                                                                                [10020:8:0ms:/home/marcus/src/gnuradio/build]
Test project /home/marcus/src/gnuradio/build
    Start 151: qa_ofdm_sync_sc_cfb
1/1 Test #151: qa_ofdm_sync_sc_cfb ..............***Failed    0.33 sec

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   0.34 sec

The following tests FAILED:
	151 - qa_ofdm_sync_sc_cfb (Failed)
Errors while running CTest
#10020|marcus>ctest -vv -R ofdm_sync                                                                                                                                [10020:8:0ms:/home/marcus/src/gnuradio/build]
Test project /home/marcus/src/gnuradio/build
    Start 151: qa_ofdm_sync_sc_cfb
1/1 Test #151: qa_ofdm_sync_sc_cfb ..............***Failed    0.32 sec

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   0.33 sec

The following tests FAILED:
	151 - qa_ofdm_sync_sc_cfb (Failed)
Errors while running CTest
#10020|marcus>ctest -vv -R ofdm_sync                                                                                                                                [10020:8:0ms:/home/marcus/src/gnuradio/build]
Test project /home/marcus/src/gnuradio/build
    Start 151: qa_ofdm_sync_sc_cfb
1/1 Test #151: qa_ofdm_sync_sc_cfb ..............   Passed    0.33 sec

100% tests passed, 0 tests failed out of 1

Total Test time (real) =   0.34 sec
#10020|marcus>ctest -vv -R ofdm_sync                                                                                                                                [10020:0:0ms:/home/marcus/src/gnuradio/build]
Test project /home/marcus/src/gnuradio/build
    Start 151: qa_ofdm_sync_sc_cfb
1/1 Test #151: qa_ofdm_sync_sc_cfb ..............***Failed    0.32 sec

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   0.33 sec

The following tests FAILED:
	151 - qa_ofdm_sync_sc_cfb (Failed)
Errors while running CTest
</pre>

jmcorgan, 2016-10-06:
I can confirm this behavior, it is quite reproducible.

noc0lour, 2018-12-13:
I'm going to hijack that issue to add it to flaky tests investigation.
```
  File "/cache/workdir/Fedora_26_0/build_Fedora_26/src/gr-digital/python/digital/qa_ofdm_sync_sc_cfb.py", line 125, in test_003_multiburst
    Detection error was: %d """ % (numpy.sum(sink_detect.data()) - n_bursts)
AssertionError: Because of statistics, it is possible (though unlikely)
that the number of detected bursts differs slightly. If the number of detects is
off by one or two, run the test again and see what happen.
Detection error was: 2 
````

PR #2232 (2018-12-16): digital: Fix ofdm_sync_sc_cfb QA flakiness
- Used a fixed random seed
- Removed messages that declare that failures aren't a big deal
- Removed some usage of channel.channel_model
- Fixed some Pylint warnings
- Factored out the creation of test vectors

Note: The SNR value for test_003_multiburst was slightly decreased, but
it's still high and given the fixed seed, it also won't matter.

Fixes #971.
closed 2018-12-16 by mbr0wn