gnuradio/gnuradio #972: FreeBSD 'libusb_get_string_descriptor'

mbr0wn, 2016-09-16:
Fresh git of HEAD before I left this morning:

[ 96%] Building C object gr-fcd/lib/CMakeFiles/gnuradio-fcd.dir/hid/hid-libusb.c.o
/root/gnuradio/gr-fcd/lib/hid/hid-libusb.c:261:19: error: static declaration of
      'libusb_get_string_descriptor' follows non-static declaration
static inline int libusb_get_string_descriptor(libusb_device_handle *dev,
                  ^
/usr/include/libusb.h:431:5: note: previous declaration is here
int     libusb_get_string_descriptor(libusb_device_handle * devh, uint8_t desc_index, uint16_...
        ^

Indeed, /usr/include/libusb.h does have a definition:

int     libusb_get_string_descriptor(libusb_device_handle * devh, uint8_t desc_index, uint16_t langid, unsigned char *data, int length);

Against this version of FreeBSD:

root@Tasha# uname -a
FreeBSD Tasha 10.2-BETA1 FreeBSD 10.2-BETA1 #0 r285434: Sun Jul 12 16:57:18 PDT 2015     root@Tasha:/disk-2/obj/disk-1/src/sys/SMUNI-FreeBSD10-amd64  amd64

The version of libusb_get_string_descriptor in libusb10_desc.c [1] is better coded than the inline where parameter checking is performed, such as checking for NULL pointers. Otherwise it's the same.

The comment in gr-fcd/lib/hid/hid-libusb.c says:

/* The FreeBSD version of libusb doesn't have this funciton. In mainline
   libusb, it's inlined in libusb.h. This function will bear a striking
   resemblence to that one, because there's about one way to code it.

The log says:

root@Tasha# svnlite log -r 277847 /usr/src/
------------------------------------------------------------------------
r277847 | hselasky | 2015-01-28 11:49:53 -0800 (Wed, 28 Jan 2015) | 4 lines

MFC r277245:
Add more USB request definitions. The values are described in section
9.4.11 and 9.4.12 of the "Universal Serial Bus 3.0 Specification"

------------------------------------------------------------------------

I don't see any great definitions against gcc5 -dM. There are:

#define __FreeBSD__ 10

Under clang there are additional definitions, such as:

#define __FreeBSD_cc_version 1000001

However that one isn't present under gcc5. 

[1] /usr/src/lib/libusb/libusb10_desc.c

mbr0wn, 2016-09-16:
Author: Michael Dickens
Date: Tue, 14 July 2015 18:44:37

mbr0wn, 2016-09-16:
Author: Michael Dickens
Date: Mon, 03 August 2015 15:08:23

marcusmueller, 2020-10-30:
gr-fcd is no longer part of GNU Radio

closed 2020-10-30 by marcusmueller