gnuradio/gnuradio #973: [pfb_clock_sync] Segfaults when taps are constant

mbr0wn, 2016-09-16:
If the filter taps are constant, the differential filter taps are zero and so its power (variable pwr).

This zero variable is used then to divide the taps for normalization:

http://gnuradio.org/redmine/projects/gnuradio/repository/revisions/master/annotate/gr-digital/lib/pfb_clock_sync_ccf_impl.cc#L310

This impacts in the variable "count" making it negative, and so d_k (phase).

Finally, the segfault pops here:

http://gnuradio.org/redmine/projects/gnuradio/repository/revisions/master/annotate/gr-digital/lib/pfb_clock_sync_ccf_impl.cc#L446

mbr0wn, 2016-09-16:
Author: Francisco Albani
Date: Thu, 16 July 2015 21:53:59

mbr0wn, 2016-09-16:
Author: Tom Rondeau
Date: Fri, 17 July 2015  0:00:29

mbr0wn, 2016-09-16:
Author: Francisco Albani
Date: Fri, 17 July 2015  0:19:26

mbr0wn, 2016-09-16:
Author: Tom Rondeau
Date: Sat, 18 July 2015  0:44:13

mbr0wn, 2016-09-16:
Author: Francisco Albani
Date: Sat, 18 July 2015  2:15:16

mbr0wn, 2016-09-16:
Author: Tom Rondeau
Date: Wed, 06 January 2016 23:11:53

marcusmueller, 2020-10-30:
fixed by Tom back in the day: we now throw an exception when normalizatio NaNs out.

closed 2020-10-30 by marcusmueller