gnuradio/gnuradio #974: qt-gui crashes on OSX 10.10

mbr0wn, 2016-09-16:
I don't see this issue on OSX 10.8, but it is repeatable about 50% of the time on 10.10. Marcus tested on his Linux (Fedora, 4.0.7-300.fc22.x86_64), and doesn't see it. But, we all see the "multiple delete" in tracebacks. It seems like a "double delete" works, while a "triple delete" does not work. Here's an example crash log from OSX 10.10:

{{{
Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       EXC_I386_GPFLT
Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   QtGui                           0x000000010c0e0967
QWidgetPrivate::close_helper(QWidgetPrivate::CloseMode) + 23
1   libgnuradio-qtgui.3.7.7.1.dylib 0x0000000114dc2595
gr::qtgui::time_sink_f_impl::~time_sink_f_impl() + 59
2   libgnuradio-qtgui.3.7.7.1.dylib 0x0000000114dc2682
gr::qtgui::time_sink_f_impl::~time_sink_f_impl() + 22
3   libgnuradio-qtgui.3.7.7.1.dylib 0x0000000114dc26d9
gr::qtgui::time_sink_f_impl::~time_sink_f_impl() + 15
4   _qtgui_swig.so                  0x0000000114c0d119
boost::detail::shared_count::~shared_count() + 45
5   _qtgui_swig.so                  0x0000000114b7aca4
_wrap_delete_time_sink_f_sptr(_object*, _object*) + 116
}}}

mbr0wn, 2016-09-16:
Author: Michael Dickens
Date: Tue, 28 July 2015 17:24:36

mbr0wn, 2016-09-16:
Author: Tom Rondeau
Date: Mon, 25 January 2016 10:36:56

mbr0wn, 2016-09-16:
Author: Michael Dickens
Date: Sun, 31 January 2016  1:09:48

mbr0wn, 2018-12-29:
@michaelld are there still known crashes of QT GUI on OS X?

michaelld, 2019-01-08:
I do experience crashes in GR, but not necessarily because of the Qt GUI. Next time I do GR testing, I'll see whether I can track down such a crash & post back here.

mbr0wn, 2019-01-08:
@michaelld If you can't find any, please close this issue.

devnulling, 2020-01-14:
@michaelld do you think is still an outstanding issue with 3.8? or should we close this issue until it comes up again?

michaelld, 2020-01-14:
IDK ... let me try & see what happens. Been a while since I used GRC in QtGUI mode ...

marcusmueller, 2020-10-30:
Heuristically, I'd say we've had heard from this in the last 9.5 months if this was an outstanding issue, plus I had students working 3.8 on their macbook

closed 2020-10-30 by marcusmueller