gnuradio/gnuradio #975: qt-gui crashes on OSX

mbr0wn, 2016-09-16:
I can replicate this issue on both 10.8 and 10.10. If I run "uhd_fft" with its new GUI / arrangement, I can coerce it into crashing by zooming in and out; eventually. Here are the outputs from 2 runs:

{{{
Using Volk machine: avx_64_mmx_orc
[UHD FFT] Setting sampling rate to: 2.0 MHz
[UHD FFT] Setting sampling rate to: 3.0 MHz
-- Asking for clock rate 48.000000 MHz... 
-- Actually got clock rate 48.000000 MHz.
-- Performing timer loopback test... pass
-- Performing timer loopback test... pass
[UHD FFT] Setting sampling rate to: 4.0 MHz
-- Asking for clock rate 32.000000 MHz... 
-- Actually got clock rate 32.000000 MHz.
-- Performing timer loopback test... pass
-- Performing timer loopback test... pass
Python(21622,0x11b6ef000) malloc: *** error for object 0x7fa160806e00:
incorrect checksum for freed object - object was probably modified after
being freed.
*** set a breakpoint in malloc_error_break to debug
Abort trap: 6
}}}

and

{{{
-- Operating over USB 3.
-- Creating ctrl transport...
-- RuntimeError: USBError -9: usb rx8 submit failed: LIBUSB_ERROR_PIPE
-- Ctrl Recv clear: LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED
-- Ctrl Send clear: LIBUSB_ERROR_NOT_FOUND
-- Data Recv clear: LIBUSB_ERROR_NOT_FOUND
-- Data Send clear: LIBUSB_ERROR_NOT_FOUND
-- Dev Reset: LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED
-- Operating over USB 3.
-- Creating ctrl transport...
-- Flusing ctrl recv buff...
-- Creating data transport...
-- Flusing data recv buff...
-- Initialize CODEC control...
-- Initialize Radio control...
-- Performing register loopback test... pass
-- Performing register loopback test... pass
-- Performing CODEC loopback test... pass
-- Performing CODEC loopback test... pass
-- Asking for clock rate 32.000000 MHz... 
-- Actually got clock rate 32.000000 MHz.
-- Performing timer loopback test... pass
-- Performing timer loopback test... pass
-- Setting master clock rate selection to 'automatic'.
-- Successfully tuned to 91.100000 MHz
-- 
Using Volk machine: avx_64_mmx_orc
objc[21612]: autorelease pool page 0x7faed481d000 corrupted
  magic 0xc2ec6344 0x4f545541 0x454c4552 0x21455341
  pthread 0x11efd4000

Illegal instruction: 4
}}}

I will attach OSX crash logs for these as well. Similar to GDB 'bt' command, with other info.

mbr0wn, 2016-09-16:
Author: Michael Dickens
Date: Tue, 28 July 2015 17:31:26

mbr0wn, 2016-09-16:
Author: Michael Dickens
Date: Wed, 29 July 2015 19:26:31

mbr0wn, 2016-09-16:
Author: Tom Rondeau
Date: Wed, 29 July 2015 20:28:02

mbr0wn, 2016-09-16:
Author: Michael Dickens
Date: Wed, 29 July 2015 20:31:36

mbr0wn, 2016-09-16:
Author: Michael Dickens
Date: Mon, 03 August 2015 20:08:36

mbr0wn, 2018-12-29:
@michaelld are there still known crashes of QT GUI on OS X?

michaelld, 2019-01-08:
hmmm ... I don't think uhd_fft crashes like this any longer. I'll test out later this week when I upgrade GR and UHD.

JayKickliter, 2019-02-20:
I can confirm this bug. I'm seeing the following when running a flowgraph with Qt GUI:

```
[bladeRF source] start: DEBUG: starting source
hijackWindow() context created for hotspot_analysis(0x7fc3e3d6da60) 1
Created Window Surface FBO QSize(2138, 1165) with samples 8
Python(69117,0x107d245c0) malloc: Incorrect checksum for freed object 0x7fc3e4187c00: probably modified after being freed.
Corrupt value: 0x7fc3c3037aa5
Python(69117,0x107d245c0) malloc: *** set a breakpoint in malloc_error_break to debug
Abort trap: 6
```

michaelld, 2019-02-20:
what version of GNU Radio / UHD / BladeRF are you running? Guessing there are printouts as each is executed.

michaelld, 2019-02-20:
Also: what version of OSX, and how is GNU Radio (etc) installed?

marcusmueller, 2020-10-30:
Closing due to lack of reproducibility

closed 2020-10-30 by marcusmueller