gnuradio/gnuradio #977: "Run To Completion" does not work

mbr0wn, 2016-09-16:
For the schematic "tst_terminate1.grc" attached the script does not
terminate after having written the desired number of samples. Consequently I
have to stop it applying "^c".

N.b. If I replace "PSK Mod" by "GMSK Mod", then the script terminates as
expected.

I also experience flow-graphs stalling, when I have hierarchically nested flow-graphs, which merge signals being produced at different sample rates and later brought to identical rate via arb-resamplers
 and I think, that this is related to the behaviour described above.

mbr0wn, 2016-09-16:
Author: Tom Rondeau
Date: Wed, 06 January 2016 22:54:19

marcusmueller, 2020-10-30:
Bug in Message Shutdown. Known, taken care of. also, PSK Mod removed.

closed 2020-10-30 by marcusmueller