gnuradio/gnuradio #978: wavfile_sink does not work properly behind a valve or selector

mbr0wn, 2016-09-16:
Connecting a signal source through a valve to a wav file sink I would expect that the samples will be stored into the wav file if the valve is open and thrown away if the valve is closed.
But if the valve is closed at the beginning, only the wav header is written to the file. After opening the valve the samples will not be stored but thrown away. After closing the flow graph the file only contains 44 bytes ( the wav file header ).

If the valve is open at the beginning the samples are stored until the valve is closed. After reopening the samples are thrown away. From this point the sink behaves like a null sink. I'd expect that the samples will be stored after reopening.

 The same happens, if you replace the valve be a selector.

Attached is a patch that fixes this bug and a flow graph ( grc ) demonstrate the problem.

mbr0wn, 2016-09-16:
Author: Marcus Müller
Date: Mon, 31 August 2015  7:32:59

mbr0wn, 2016-09-16:
Author: Martin Braun
Date: Mon, 31 August 2015 17:07:44

mbr0wn, 2016-09-16:
Author: Marcus Müller
Date: Mon, 31 August 2015 17:13:23

mbr0wn, 2016-09-16:
Author: Martin Braun
Date: Thu, 03 September 2015 16:44:50

mbr0wn, 2016-09-16:
Author: Tom Rondeau
Date: Wed, 30 December 2015 17:24:01

friedrich9, 2019-07-04:
hello!
I am not used to work with this platform.
From where I can download your patch?
Thank you!

dkozel, 2019-07-06:
Hello @friedrich9 
This bug and related posts were imported from our previous bug tracker. The patch is likely lost now as the old bug tracker has been shutdown. Are you encountering this problem? There has been a lot of development since this bug was initially reported.

marcusmueller, 2020-10-30:
all three elements in this issue have been completely rewritten or removed. Closing!

closed 2020-10-30 by marcusmueller