gnuradio/gnuradio #979: Half-band equiripple FIR filter (and others) design fails

mbr0wn, 2016-09-16:
In the GR filter design tool, design of half-band equiripple FIR filters fails dues to a namespace error in gr-filter/python/filter/design/fir_design.py.

design_opt_hb() should specify "mainwin.EQUIRIPPLE_FILT" vs. "self.EQUIRIPPLE_FILT" in its return parameters.

Incorrect:
https://github.com/gnuradio/gnuradio/blob/master/gr-filter/python/filter/design/fir_design.py#L380

Correct:
https://github.com/gnuradio/gnuradio/blob/master/gr-filter/python/filter/design/fir_design.py#L351

There are actually 3 instances of "self.EQUIRIPPLE_FILT" in the file that need to be fixed.

The tool emits this traceback when it trips across it the bug:
Traceback (most recent call last):
  File "/home/andy/work/target/lib64/python2.7/site-packages/gnuradio/filter/filter_design.py", line 867, in design
    self.design_fir(ftype, fs, gain, winstr)
  File "/home/andy/work/target/lib64/python2.7/site-packages/gnuradio/filter/filter_design.py", line 889, in design_fir
    taps,params,r = designer[ftype](fs, gain, self)
  File "/home/andy/work/target/lib64/python2.7/site-packages/gnuradio/filter/fir_design.py", line 380, in design_opt_hb
    params = {"fs": fs, "gain": gain, "wintype": self.EQUIRIPPLE_FILT,
NameError: global name 'self' is not defined

mbr0wn, 2016-09-16:
Author: Sreeraj Rajendran
Date: Mon, 07 September 2015  9:13:13

argilo, 2020-07-06:
The three instances of `self.EQUIRIPPLE_FILT` were corrected to `mainwin.EQUIRIPPLE_FILT` in #2495, and I verified that a half-band equiripple FIR filter can now be generated. So it looks like this can be closed off.

closed 2020-07-06 by argilo