gnuradio/gnuradio #980: Tag propagation and stream_to_vector

mbr0wn, 2016-09-16:
block_executor.cc/propagate_tags() handles offsets on rate changing blocks like this:

new_tag.offset = ((double)new_tag.offset * rrate) + 0.5;

For the stream_to_vector block (and probably others), this potentially assigns tags from one group of input items to two different output items. The behavior is correct for streams, but doesn't account for groupings of items.

Original problem reported by David Halls.

mbr0wn, 2016-09-16:
Author: Jeff Long
Date: Thu, 10 September 2015 23:20:08

mbr0wn, 2016-09-16:
Author: Tom Rondeau
Date: Wed, 30 December 2015 17:22:08

mbr0wn, 2016-09-16:
Author: Tom Rondeau
Date: Wed, 30 December 2015 21:50:25

mbr0wn, 2016-09-16:
Author: Jeff Long
Date: Fri, 08 January 2016  0:53:28

mbr0wn, 2016-09-16:
Author: Jeff Long
Date: Mon, 18 January 2016 21:19:11

marcusmueller, 2020-10-30:
Tracked elsewhere

closed 2020-10-30 by marcusmueller