gnuradio/gnuradio #981: Stream tags getting lost when changing sample rates in a block

mbr0wn, 2016-09-16:
For a test case, use the fractional_resampler. When starting with a sample rate change of 2, the relative rate is 2. Input tags at samples:

In Offset: 0
In Offset: 100
....
In Offset: 1000

Will be moved to the output at samples

Out Offset: 0
Out Offset: 200
....
Out Offset: 2000

Because: Out Offset = relative_rate * In Offset

The fractional_resampler has now processed 2000 samples into its output port (nitems_written(0) -> 2000)

Change relative_rate to 1 now.

In Offset: 1100
In Offset: 1200
....
In Offset: 2000

Out Offset: 1100
Out Offset: 1200
....
Out Offset: 2000

But we have already passed this value in the output buffer based on nitems_written(0). We've added another 1000 items, so nitems_written(0) -> 3000, instead, because it's a simple counter.

We need to either reset the counters when we change the sample rates on these blocks or account for the change in offsets that occur because of our counting.

mbr0wn, 2016-09-16:
Author: Tom Rondeau
Date: Fri, 25 September 2015 21:28:23

marcusmueller, 2020-10-30:
we've had a rework of resampling blocks. It was a big thing - this should be fixed now.

closed 2020-10-30 by marcusmueller