gnuradio/gnuradio #1000: Gnuradio fails to compile with cmake-3.5

mbr0wn, 2016-09-17:
The problem is that gnuradio uses CMAKE_SOURCE_DIR/CMAKE_BINARY_DIR which doesn't seem to be ever supported by cmake and now appears to be no longer allowed. The fix is probably to change it to PROJECT_SOURCE_DIR/PROJECT_BINARY_DIR. The original downstream report:

https://bugzilla.redhat.com/show_bug.cgi?id=1311358

Github pull request:
https://github.com/gnuradio/volk/pull/65

mbr0wn, 2016-09-17:
Author: Michael Dickens
Date: Wed, 24 February 2016 17:08:57

I don't know what version of CMake the OP is using, but CMAKE_SOURCE_DIR and CMAKE_BINARY_DIR have been part of CMake since at least 2.6. Their meaning has not changed between versions. See also:[https://cmake.org/cmake/help/cmake2.6docs.html#variable:CMAKE_SOURCE_DIR][1] [https://cmake.org/cmake/help/v2.8.12/cmake.html#variable:CMAKE_SOURCE_DIR][2] [https://cmake.org/cmake/help/v3.5/variable/CMAKE_SOURCE_DIR.html][3] 

    

  [1]: https://cmake.org/cmake/help/cmake2.6docs.html#variable:CMAKE_SOURCE_DIR
  [2]: https://cmake.org/cmake/help/v2.8.12/cmake.html#variable:CMAKE_SOURCE_DIR
  [3]: https://cmake.org/cmake/help/v3.5/variable/CMAKE_SOURCE_DIR.html

chevdor, 2016-12-19:
Is there any news on this issue?
We are at cmake 3.7.1 and it seems that grc is stuck behind at some 3.3.2

jmcorgan, 2016-12-19:
@mbr0wn @michaelld @n-west I'd like a summary of where this stands--what is the delta between today's VOLK and gnuradio codebase and what needs to exist to fix this problem in a proper and complete way.

n-west, 2016-12-19:
I'm not clear what the problem is. I've used cmake 3.5 and then some to build gnuradio + volk (and in fact do so often). I even merged the PR that looked harmless, but supposedly fixed the issue.

@chevdor what do you mean grc is stuck at 3.3.2? Are you saying that your distro isn't providing a more recent version or that you can't build a more recent version?

chevdor, 2016-12-19:
@n-west I am using OSX Sierra. I have cmake 3.7.1 but gnuradio won´t compile with anything above 3.3.2
That would explain the issue with 3.5

jmcorgan, 2016-12-19:
We routinely compile with CMake 3.5.1 on Linux, so there must be a more specific issue.

michaelld, 2016-12-20:
@chevdor: I'm running macOS Sierra (10.12) and use the latest CMake (3.7 snapshot g08aac from 20161213; 3.7.1 works about as well for GR needs in my testing). I have no issues building or executing GNU Radio. How is GR installed? How is CMake installed? If you're not using homebrew or MacPorts, I'd recommend moving to one of them since they make these installations so much easier. I'm biased toward MacPorts, but I know plenty of folks who prefer homebrew; either works well for the vast majority of end-users.

chevdor, 2016-12-20:
@michaelld thanks for your feedback. The fact that it runs for you is encouraging.

I am using homebrew as much as I can. This is how I installed cmake 3.7.1
The difficulty atm with homebrew is that qt(4) is being deprecated but there is no official gnuradio tag for qt5 (yet). @ilovezfs made a formula (homebrew) for qt5 and I could build grc successfully using it. This formula uses cmake 3.3.2 btw.

I then built gr-osmosdr from source. Yesterday, I even rebuilt gt5 using homebrew.

At that point, I can start grc but it does not work properly.
I am testing with a simple workflow as shown below:
[image]

Here is the result of running this workflow:
```
Generating: '/Users/chevdor/.../gnuradio-homebrew/top_block.py'

Executing: /usr/local/opt/python/bin/python2.7 -u /Users/chevdor/.../gnuradio-homebrew/top_block.py

Traceback (most recent call last):
  File "/Users/chevdor/.../gnuradio-homebrew/top_block.py", line 21, in <module>
    from PyQt5 import Qt
ImportError: dlopen(/usr/local/opt/pyqt5/lib/python3.5/site-packages/PyQt5/Qt.so, 2): Symbol not found: _PyModule_Create2
  Referenced from: /usr/local/opt/pyqt5/lib/python3.5/site-packages/PyQt5/Qt.so
  Expected in: flat namespace
 in /usr/local/opt/pyqt5/lib/python3.5/site-packages/PyQt5/Qt.so

>>> Done (return code 1)
```

michaelld, 2016-12-20:
@jmcorgan : GNU Radio does not yet build/work with Python 3.5 ... am I correct?

michaelld, 2016-12-20:
@chevdor : I note that you're using the python2.7 executable, but PyQt5 is installed in the python3.5 directory ... seems like a mismatch there! Maybe get PyQt5 installed for Python 2.7 & see if that works? I've found that mixing different installs of the same Python (e.g., Apple 2.7 and MacPorts 2.7) can be made to work ... but, mixing between Python versions does not generally work because of API and ABI incompatibilities.

jmcorgan, 2016-12-20:
Not yet.  The next branch (that will become 3.8) will eventually support Python >= 3.4.

michaelld, 2016-12-20:
OK. Thx.

ilovezfs, 2016-12-20:
which version does `next` currently support?

jmcorgan, 2016-12-20:
Also, we don't have an "official" tag on next branch as it is going through some deep changes soon and will likely be operating in "degraded" mode until the bugs are worked out.

jmcorgan, 2016-12-20:
Currently, the next branch supports python2 2.7 or greater, but not yet python3.

chevdor, 2016-12-20:
After quite some fighting, it seems I am on a good way. If the current compile works, that will confirm that what I said previously is no longer correct and that cmake 3.7.1 does work (which would be great).

@michaelld indeed, I saw that I had pyqt5 installed but only with python3 support. I reinstalled it with homebrew with `--with-python` and it seems to work so far.

Just for the record, cmake kept on trying to find volk one directory higher than it really was. So I ended up compiling volk separately (which is very easy) and then used `cmake -Wno-dev -DENABLE_INTERNAL_VOLK=OFF ..` to build gnuradio without rebuilding volk.

I am using the branch next-qt5 btw.

chevdor, 2016-12-20:
I finally could build and I am probably now hitting issues due to the fact that the branch next-qt5 is not ready yet.
I tested another simple flow:
[image]

and got:
```
Generating: '/Users/chevdor/projects/rf/gnuradio/build/top_block.py'

Executing: /usr/local/opt/python/bin/python2.7 -u /Users/chevdor/projects/rf/gnuradio/build/top_block.py

Traceback (most recent call last):
  File "/Users/chevdor/projects/rf/gnuradio/build/top_block.py", line 142, in <module>
    main()
  File "/Users/chevdor/projects/rf/gnuradio/build/top_block.py", line 130, in main
    tb = top_block_cls()
  File "/Users/chevdor/projects/rf/gnuradio/build/top_block.py", line 55, in __init__
    self.restoreGeometry(self.settings.value("geometry"))
TypeError: restoreGeometry(self, Union[QByteArray, bytes, bytearray]): argument 1 has unexpected type 'NoneType'

>>> Done
```

The key points to build from source (branch **next-qt5**) based on a homebrew install are:
- ensure qt5 is installed and qt(4) is NOT
- ensure pyqt5 is installed with `--with-python` (at least until grc supports python3)
- build/install volk manually/separately (otherwise cmake searches for it in the wrong location)
- ensure to install all the deps (no surprise here, just don´t install sphinx from homebrew): pip install mako sphinx cython pycomedi

Use the following cmake:
`cmake -Wno-dev -DENABLE_INTERNAL_VOLK=OFF -DPYTHON_EXECUTABLE=/usr/local/bin/python -DPYTHON_INCLUDE_DIR=/usr/local/Frameworks/Python.framework/Versions/2.7/Headers -DPYTHON_LIBRARY=/usr/local/Frameworks/Python.framework/Versions/2.7/Python -DGR_PYTHON_DIR=/usr/local/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages ..
`

jmcorgan, 2016-12-20:
The next-qt5 branch has already been merged into next and is no longer in our repository, and the qt4/qt5 issue has already been fixed. I'm not sure the issue about volk needing to be compiled separately; it works find as an embedded cmake with Ubuntu and Fedora linux flavors.

If you're still using next-qt5 then that might explain some of the things that you are seeing.

chevdor, 2016-12-20:
@jmcorgan ok, yes, I was still on next-qt5. I will try with the **next** branch then.
The volk issue is no blocker with the info above. cmake just look for volk in '..' instead of '../volk' on OSX.

jmcorgan, 2016-12-20:
Also, I the -Wno-dev issue has been corrected, not sure if before or after next-qt5 merge, but you shouldn't need that.