gnuradio/gnuradio #1007: pmt double serialization broken

mbr0wn, 2016-09-17:
PMT does not correctly serialize and deserialize double.

Consider the following example to reconstruct the issue:

import pmt
pmt.to_python(pmt.deserialize_str(pmt.serialize_str(pmt.to_pmt(2370e9/3))))
> 789999976448.0
pmt.to_python(pmt.to_pmt(2370e9/3))
> 790000000000.0

Obviously the numbers are not the same.

mbr0wn, 2016-09-17:
Author: Martin Braun
Date: Fri, 17 June 2016 19:07:03

Did this get resolved in [https://github.com/gnuradio/gnuradio/commit/cd07e5e56647a9380ccd0f25960164d72b338e10#diff-960cadcf7ef5dc2327e5200893d43163][1] ?

    

  [1]: https://github.com/gnuradio/gnuradio/commit/cd07e5e56647a9380ccd0f25960164d72b338e10#diff-960cadcf7ef5dc2327e5200893d43163

mbr0wn, 2016-09-17:
Author: Martin Braun
Date: Fri, 17 June 2016 19:07:24

- **Status** changed from New to Feedback - **Assignee** set to Johnathan Corgan - **Resolution** set to fixed

mbr0wn, 2016-09-17:
Author: Johannes Schmitz
Date: Mon, 20 June 2016 11:40:21

Martin Braun wrote:

> Did this get resolved in [https://github.com/gnuradio/gnuradio/commit/cd07e5e56647a9380ccd0f25960164d72b338e10#diff-960cadcf7ef5dc2327e5200893d43163][1] ?

	The example given in the description seems to be running fine now with this patch.

	However, we are suspecting some additional similar problem with pmt serialization in zeromq blocks somewhere.We are still investigating. I guess we can probably handle this as an additional issue and keep this one closed for the moment.

    

  [1]: https://github.com/gnuradio/gnuradio/commit/cd07e5e56647a9380ccd0f25960164d72b338e10#diff-960cadcf7ef5dc2327e5200893d43163

mbr0wn, 2016-09-17:
Author: Johannes Schmitz
Date: Mon, 20 June 2016 15:36:50

Johannes Schmitz wrote:

> However, we are suspecting some additional similar problem with pmt serialization in zeromq blocks somewhere.We are still investigating. I guess we can probably handle this as an additional issue and keep this one closed for the moment.

	We found the remaining problem on our side. In Python we where using a to_float instead of to_python or to_double.So the issue can be closed I suggest.

jmcorgan, 2016-09-18:
The original author confirmed this was fixed.

closed 2016-09-18 by jmcorgan