gnuradio/gnuradio #1010: No QA test for gr::prefs / config file parser

mbr0wn, 2016-09-17:
There's no QA for gr::prefs _at all_.

Being core functionality, http://gnuradio.org/redmine/issues/918 will have to wait until I implement one.

marcusmueller, 2020-10-30:
> There's no QA for gr::prefs _at all_.

Not true:
```
    def test_001(self):
        p = gr.prefs()

        # Read some options
        self.assertFalse(p.has_option('doesnt', 'exist'))

        # At the time these tests are run, there is not necessarily a default
        # configuration on the build system, so not much to do with testing here
```

But truth be told, that is not *sufficient* testing.

marcusmueller, 2020-10-30:
This being a really well-stressed multi-platform-problematic core component, lack of testing *does* constitute a bug, since we'll not notice failures.

There's been at least one acute instance of this biting us (namely, 3.7 < 3.7.13 can't change audio settings since the prefs file handling was broken).

marcusmueller, 2023-12-20:
remains valid, should be sensibly fixable, long-term maintainability issue.