gnuradio/gnuradio #1011: Flowgraph validation with bypassed blocks

mbr0wn, 2016-09-17:
flow graph validation is not working correctly on a chain of bypassed and disabled blocks:

Connect a source, some proc block and a sink. 
Disable the sink and bypass the proc block.

-> since the connection from the source now goes nowhere it should show an error but doesn't.

To fix this, we could make use of the code used for virtual connections. 
However, that code is pretty messy, so maybe we refactor it first.

marcusmueller, 2020-10-30:
@skoslowski We'd need a quick assessment of this. I think it's solved, but I'm not sure whether that's just the case for the trivial case (i.e. no dependencies of dependencies). If this is actually solved, or no longer applicable, please do just close.

marcusmueller, 2023-10-24:
Problem persists as of main today
[image]