gnuradio/gnuradio #1015: Make rotator_cc more convenient to use

mbr0wn, 2016-09-17:
In order to frequency-shift a signal there are three reasonable options available: building a quadrature mixer from primitives (signal source ? multiply), using the @freq_xlating_fir_filter@ block, or using the @rotator_cc@ block. Assuming one does not wish to filter at the same time, the most appropriate block is @rotator_cc@. However, it is notably harder to use, in typical cases, than the other options, because it does not take its parameter in Hz but rather in radian/s, which, while a natural unit for the mathematical operation, is not a natural unit for basic software radio applications, and is not even very often used elsewhere in GNU Radio.

The conversion in Python is @(2 * math.pi) * (shift / rate)@ which is not that much code, but it is not _obviously the correct calculation at a glance_ (important when debugging ?why I am I getting nonsense out?), and in GRC it requires either adding a block to @import math@ or entering a value for pi.

In order to improve this situation, I propose the following two changes which would allow easier use of the block (yet without hiding the details):

# Add a static member function @gr::blocks::rotator_cc::phase_inc(double shift_freq, double sampling_freq)@ which performs the above calculation. This is useful for hand-written code.
# In @blocks_rotator_cc.xml@, add an option which can be set to allow input of the two frequencies rather than the phase increment; I believe this is feasible by using the @<hide>@ element on the non-applicable fields. This option would change the generated code to use the above @phase_inc@ function, but not be itself passed to the rotator block.

These facilities could also of course be used to express a frequency shift in fractions of the sampling rate by letting @sampling_freq = 1@.

I believe it is worth adding these features because it will help people who are just beginning to learn SDR/DSP by increasing consistency: from the viewpoint of a GNU Radio Companion user, nearly all blocks that work with frequencies take a sampling rate parameter and frequency values in Hz, and @rotator_cc@ is an exception.

mbr0wn, 2016-09-17:
Author: Martin Braun
Date: Tue, 24 May 2016 18:04:42

- **Status** changed from New to Feedback
I second this. I'm not entirely convinced about the phase_inc() method, but what the hell, it's a one-liner and it won't blow up mem usage or anything if it's a static func.

	The GRC changes would definitely be a great idea, no doubt about it.

	Kevin, would you be up to doing those changes? I see Marcus is assigned, but why?

	On a side note, for beginner tutorials, depending on the goal of the tutorial, doing the siggen -> multiplier chain might actually be clearer.

mbr0wn, 2016-09-17:
Author: Marcus Müller
Date: Tue, 24 May 2016 18:18:35

I'm assigned because I asked to be; this comes from a discussion on dsp.stackexchange or stackoverflow.com (that I of course can't find anymore now that I try). Point was that the Rotator block is hard to use for beginners, and this would make it easier.

	I asked Kevin to create an issue. Not wanting to let that float without Assignee, I also asked him to assign it to me. If anyone else wants to implement this, he can be assignee, too!

	Also, now that I have my favourite SDR PhD in here, I'd also like to propose changing the (GRC-displayed) Name from "Rotator" to "Frequency Shift (Complex Phase Rotator)".

	Awalls rightfully mentions that the same usability problem applies to "frequency modulator" and "quadrature demodulator".

mbr0wn, 2016-09-17:
Author: Kevin Reid
Date: Wed, 25 May 2016  0:55:34

I also like the idea of renaming the block in GRC — I thought of it myself but decided not to mention it to keep my proposal focused. Note that that rename would be inconsistent with the filter-and-shift blocks named "Frequency Xlating...". (Which is not a naming precedent I like, but it does match the block class name, loosely. Though now that GRC can show generated code, matching names are not as useful as they were.)

smunaut, 2016-09-21:
I wouldn't remove the old name / behavior ... sometime you want things to be purely analytical and sample_rate independent. (I know I definitely have some flowgraph that don't care about absolute sample rate and work purely as relative one and using radian / sample etc ...)

You can simply add an overloaded constructor with 2 doubles to support the shift, samplerate use case and add an aliased GRC block for it with those params. This way both use case are supported and it doesn't break anything.

ghostop14, 2020-01-15:
Just happened to see this.  PR #3046 adds an additional convenience wrapper called Frequency Shift that would take frequency and sample rate to do the math for the shift.  I'd add it as a new block to not break backward compatibility for folks using a rotator in a flowgraph already.

rear1019, 2020-03-18:
Closing, fixed by PR #3046. We might want to backport this to maint-3.8

closed 2020-03-18 by rear1019