gnuradio/gnuradio #1019: generic volk_32fc_s32fc_x2_rotator_32fc does not normalize rotator on short sequences

mbr0wn, 2016-09-17:
In gnuradio/volk/kernels/volk/volk_32fc_s32fc_x2_rotator_32fc.h, all special functions are normalizing rotator to unit vector at the end:

<pre>
    if (i) { 
        tmp1 = _mm_mul_ps(phase_Val, phase_Val);
        tmp2 = _mm_hadd_ps(tmp1, tmp1);
        tmp1 = _mm_shuffle_ps(tmp2, tmp2, 0xD8);
        tmp2 = _mm_sqrt_ps(tmp1);
        phase_Val = _mm_div_ps(phase_Val, tmp2);
    }

</pre>

However, the generic implementation normalizes it only every ROTATOR_RELOAD (currently 512) samples. This means that repeated rotations of less than 512 samples may lead to walking into infinity or zero.

Code to reproduce:

<pre>
int main(int argc, char **argv) {

  int N = 100;
  unsigned int alignment = volk_get_alignment();
  lv_32fc_t* in  = (lv_32fc_t*)volk_malloc(sizeof(lv_32fc_t)*N, alignment);
  lv_32fc_t* out = (lv_32fc_t*)volk_malloc(sizeof(lv_32fc_t)*N, alignment);
  float frequency = 0.1f;
  lv_32fc_t phase_increment = lv_cmake(cos(frequency), sin(frequency));
  lv_32fc_t phase = lv_cmake(1.f, 0.0f);

  for(int i = 0; i<1000*1000*1000; i++) {
    volk_32fc_s32fc_x2_rotator_32fc(out, in, phase_increment, &phase, N);
    if(i%1000000 == 0) {
      printf("at %i vector magnitude: %f\n", i, hypotf(lv_creal(phase), lv_cimag(phase)));
    }
  }

}
</pre>

AVX impl:

<pre>
at 0 vector magnitude: 1.000000
at 1000000 vector magnitude: 1.000000
at 2000000 vector magnitude: 1.000000
at 3000000 vector magnitude: 1.000000
at 4000000 vector magnitude: 1.000000
at 5000000 vector magnitude: 1.000000
at 6000000 vector magnitude: 1.000000
at 7000000 vector magnitude: 1.000000
at 8000000 vector magnitude: 1.000000
^C
</pre>

generic impl:

<pre>
at 0 vector magnitude: 1.000001
at 1000000 vector magnitude: 3.445564
at 2000000 vector magnitude: 11.871789
at 3000000 vector magnitude: 40.909409
at 4000000 vector magnitude: 140.926544
at 5000000 vector magnitude: 485.528656
at 6000000 vector magnitude: 1673.781860
at 7000000 vector magnitude: 5765.379395
at 8000000 vector magnitude: 19869.707031
^C
</pre>

I suggest the atatched patch.

mbr0wn, 2016-09-17:
Author: Jan Hrach
Date: Fri, 29 July 2016 22:32:19

The attachements (and previews) unfortunately don't seem to work. The patch inserted inline:

diff --git a/kernels/volk/volk_32fc_s32fc_x2_rotator_32fc.h b/kernels/volk/volk_32fc_s32fc_x2_rotator_32fc.h
index 70bc153..76ed730 100644
--- a/kernels/volk/volk_32fc_s32fc_x2_rotator_32fc.h
+++ b/kernels/volk/volk_32fc_s32fc_x2_rotator_32fc.h
@@ -109,6 +109,14 @@ static inline void volk_32fc_s32fc_x2_rotator_32fc_generic(lv_32fc_t* outVector,
         *outVector++ = *inVector++ * (*phase);
         (*phase) *= phase_inc;
     }
+    if(i) {
+#ifdef __cplusplus
+        (*phase) /= std::abs((*phase));
+#else
+        (*phase) /= hypotf(lv_creal(*phase), lv_cimag(*phase));
+#endif
+
+    }

 }

devnulling, 2020-01-21:
@michaelld can you verify if this is still an open issue? maybe this should be moved over to the volk repo issue list?

michaelld, 2020-01-21:
@devnulling looks like it's still an issue:
```
% VOLK_GENERIC=1 ./test_volk_32fc_s32fc_x2_rotator_32gc_issue_1019
at 0 vector magnitude: 1.000001
at 1000000 vector magnitude: 3.445564
at 2000000 vector magnitude: 11.871789
at 3000000 vector magnitude: 40.909409
at 4000000 vector magnitude: 140.926544
at 5000000 vector magnitude: 485.528656
at 6000000 vector magnitude: 1673.781860
^C

% ./test_volk_32fc_s32fc_x2_rotator_32gc_issue_1019
at 0 vector magnitude: 1.000000
at 1000000 vector magnitude: 1.000000
at 2000000 vector magnitude: 1.000000
at 3000000 vector magnitude: 1.000000
at 4000000 vector magnitude: 1.000000
at 5000000 vector magnitude: 1.000000
^C
```

michaelld, 2020-01-21:
Moving this issue to Volk.

closed 2020-01-21 by michaelld