gnuradio/gnuradio #1020: Fix msk_timing_recovery_cc time_est value handling

mbr0wn, 2016-09-17:
The @digital::msk_timing_recovery_cc@ block needs to handle all @time_est@ values properly, including garbage values. Currently if a @time_est@ tag's value (e.g., from upstream @digital::corr_est_cc@ block) is outside some acceptable range, it causes the @filter::mmse_fir_interpolater_cc@ block to throw an error:

<code>thread[thread-per-block[17]: <block msk_timing_recovery_cc (10)>]: mmse_fir_interpolator_cc: imu out of bounds</code>

See this discussion for additional details: https://lists.gnu.org/archive/html/discuss-gnuradio/2016-03/msg00705.html

jmcorgan, 2016-11-04:
@mbr0wn Are you working on this?  Should we aim to include a fix in 3.7.10.2?

bistromath, 2016-11-04:
Aw, this one is my fault (although I think the problem is shared in the M&M block as well), so I'll fix it.

bistromath, 2016-11-04:
You want it based off maint?

jmcorgan, 2016-11-04:
Yes.

bistromath, 2016-11-04:
#1084 

Just made it yell with GR_LOG_WARN and ignore the tag if it's outside (-1, 1). I _could_ treat it like a valid tag and offset the index by the integer part, but I don't think the timing recovery block should be responsible for making educated guesses based on incorrectly-generated tags. We should fix corr_est so it doesn't issue these tags as well, which the WARN text will hopefully remind us to do.

jmcorgan, 2016-11-04:
Fixed in #1084

closed 2016-11-04 by jmcorgan