gnuradio/gnuradio #1021: Fix digital::corr_est_cc block time_est values to be in range [-0.5, 0.5]

mbr0wn, 2016-09-17:
The @digital::corr_est_cc@ block claims to generate @time_est@ tags with values in the range [-0.5, 0.5], which is supposed to be a fractional sample delay estimate, but it sometimes yields numbers outside this range. It has been observed outputting numbers like -1.3, -1.7, etc., which is causing @digital::msk_timing_recovery_cc@ to crash (although that block needs to do its own bounds checking too; see #929).

See this discussion for additional details: https://lists.gnu.org/archive/html/discuss-gnuradio/2016-03/msg00705.html

mbr0wn, 2016-09-17:
Author: Sean Nowlan
Date: Fri, 05 August 2016 20:24:10

Note that I was on v3.7.9.2 when I experienced this error. It may have been fixed as a side effect of replacing parabolic interpolation with linear interpolation in this commit: [https://github.com/gnuradio/gnuradio/pull/853/commits/32d1c7d3188bf3107b45f350e44907cf23faee3e][1] 

    

  [1]: https://github.com/gnuradio/gnuradio/pull/853/commits/32d1c7d3188bf3107b45f350e44907cf23faee3e

jmcorgan, 2016-09-18:
This was fixed indeed fixed in 32d1c7d3.

closed 2016-09-18 by jmcorgan