gnuradio/gnuradio #984: fm_emph.py bad coefficients

mbr0wn, 2016-09-17:
In line 62 the ataps coefficients are:
ataps = [1, a1]

This is wrong it should be:
ataps = [1, -a1]

If you represent the frequency response with the above filter you will see a flat response with a high attenuation for 
all frequencies

mbr0wn, 2016-09-17:
Author: Alberto Albiol
Date: Wed, 11 November 2015  9:31:34

Alberto Albiol wrote:

> In line 62 of fm_emph.py the ataps coefficients are:ataps = [1, a1]

	This is wrong it should be:ataps = [1, -a1]

	If you represent the frequency response with the above filter you will see a flat response with a high attenuation for all frequencies

mbr0wn, 2016-09-17:
Author: Tom Rondeau
Date: Wed, 30 December 2015 18:01:54

- **Category** set to gr-analog - **Status** changed from New to Feedback - **Assignee** set to Tom Rondeau
Alberto,

	Can you take a look at the docs for the IIR filter in use here:[http://gnuradio.org/doc/doxygen/classgr_1_1filter_1_1iir__filter__ffd.html][1] 

	Specifically, there's an "oldstyle" flag that defaults to true, so is in use in the fm_emph.py file, that inverts the sign of the coefficient. I believe that this means that the code is correct as is but would be incorrect if we changed oldstyle=False.

	If I'm correct, then I would be in favor of updating the taps to what you suggest and switching this flag to False. Eventually, we want to get rid of this "oldstyle" concept and focus on a single way of representing the IIR filter taps.

    

  [1]: http://gnuradio.org/doc/doxygen/classgr_1_1filter_1_1iir__filter__ffd.html

mbr0wn, 2016-09-17:
Author: Alberto Albiol
Date: Thu, 31 December 2015 11:26:23

- **File** [Screen Shot 2015-12-31 at 12.04.59.png][1]  added - **File** [Screen Shot 2015-12-31 at 12.04.59.png][2]  added
Tom,Effectively this old/new style issue is also very confusing for my students.

In this case we can fix the issue by: 
	
	 - Option 1
	

	    ataps=[1, -a1] , oldstyle=True

	Option 2

	    ataps=[1, a1] (as it is right now),  oldstyle=False

	These two solutions are equivalen. I think that second option is more intuitive though.

	I also agree that this old/new style is also confusing and it would bemuch more intuitive if coefficients are passed in an unique way (matlab style for instance).

	In any case you can also check that the frequency response is correct by plotting it. I attach the right and wrong responses that you get, notice that as expected the right response is a low pass filter

    

  [1]: /redmine/attachments/download/926/Screen%20Shot%202015-12-31%20at%2012.04.59.png
  [2]: /redmine/attachments/download/927/Screen%20Shot%202015-12-31%20at%2012.04.59.png

mbr0wn, 2016-09-17:
Author: Alberto Albiol
Date: Thu, 31 December 2015 11:29:13

- **File** [Screen Shot 2015-12-31 at 12.05.44.png][1]  added
Alberto Albiol wrote:

> In line 62 the ataps coefficients are:ataps = [1, a1]

	This is wrong it should be:ataps = [1, -a1]

	If you represent the frequency response with the above filter you will see a flat response with a high attenuation for all frequencies

    

  [1]: /redmine/attachments/download/928/Screen%20Shot%202015-12-31%20at%2012.05.44.png

mbr0wn, 2016-09-17:
Author: Tom Rondeau
Date: Thu, 31 December 2015 17:52:04

- **File** [nbfm_sim.grc][1]  added
I agree that the "oldstyle" flag can be confusing. The problem is that we used the IIR filters this way for a while but implemented the "new style" to make things easier in the future. We wanted people to be more comfortable with the way IIR filter taps were designed and used here, but simply changing this behavior meant that anyone using IIR filters in GNU Radio would suddenly have their applications stop working correctly.

	We introduced the "oldstyle" flag to get around this problem as a temporary fix. Eventually, we will remove the flag and move the IIR filters to the new style only. This will likely happen for 3.8.

	In the meantime, because of this confusion, I'm not entirely convinced what the right answer is. I'd like to see an end-to-end solution work properly (see the attached example). I think a serious problem here is that we don't actually have a pre-emphasis filter designed. You can see the taps in fm_emph.py's fm_preemph are just [1] and [1].

    

  [1]: /redmine/attachments/download/929/nbfm_sim.grc

mbr0wn, 2016-09-17:
Author: Alberto Albiol
Date: Fri, 08 January 2016 11:48:52

- **File** [preemphasis_filter.pdf][1]  added
Dear TomAside from the detail of the "oldstyle" flag which is a minor error, I have been working these days towards designing  a pre-emphasis filter. I have attached a file with the details (and matlab code) so you can use it to update the pre-empashis filter. Keep in mind these points:1- There is a minor error in the prewarping expression used in the deemphasis filter (explained in the attached file)2- Always keep the sampling frequency high to prevent the nonlinear frequency warping of the bilinear transformation 3- The frequency response of the analog preemphasis filter should be slightly modified to avoid unstability during the bilinear transformation (again see the document for details)

	I hope this will help

    

  [1]: /redmine/attachments/download/941/preemphasis_filter.pdf

awalls-cx18, 2016-09-27:
The following pull request, which is already merged, fixed the FM pre-emphasis and de-emphasis filters:
https://github.com/gnuradio/gnuradio/pull/777
#777

awalls-cx18, 2016-09-27:
@mbr0wn github unassigned trondeau by me just making a comment.  That seems wrong.

mbr0wn, 2016-09-27:
Hm that shouldn't happen... then again @trondeau should probably have been unassigned so this is fine.

closed 2016-09-27 by jmcorgan