gnuradio/gnuradio #985: GRC Bus Connections

mbr0wn, 2016-09-17:
Messages and stream sink ports on the same block do not work well with GRC bus ports. A good example is the QT Frequency sink with 8 stream ports; since there is a hidden message port, GRC creates 9 port bus connection rather than the expected 8.

mbr0wn, 2016-09-17:
Author: Sebastian Koslowski
Date: Fri, 20 November 2015  7:45:12

Can't say I am surprised. The code for Bus Ports is a mess on so many levels. I recommend refactoring it first. I had started on this at some time ago (its on my github). Maybe that helps.

mbr0wn, 2016-09-17:
Author: Sean Nowlan
Date: Fri, 18 March 2016 23:04:56

An example of this bug can be seen when trying to load a flowgraph [1] from Tom's PFB channelizer/synthesizer blog post [2] on 3.7.9.1.

	[1] [https://static.squarespace.com/static/543ae9afe4b0c3b808d72acd/543aee1fe4b09162d08633d9/543aee20e4b09162d0863550/1395274396537/fm_channelize_and_view.grc][1] [2] [http://www.trondeau.com/examples/2014/1/23/pfb-channelizers-and-synthesizers.html][2] 

    

  [1]: https://static.squarespace.com/static/543ae9afe4b0c3b808d72acd/543aee1fe4b09162d08633d9/543aee20e4b09162d0863550/1395274396537/fm_channelize_and_view.grc
  [2]: http://www.trondeau.com/examples/2014/1/23/pfb-channelizers-and-synthesizers.html

kb1ca, 2019-06-08:
For others who might be Googling; the message generated when this issue is encountered is "Error: port connections must have the same cardinality". 

It still seems to be an issue with 3.7.13.5 when trying to work the Polyphase Channelizer example flowgraphs.

marcusmueller, 2023-10-24:
This is fixed in main, the bus ports only affect streams

closed 2023-10-24 by marcusmueller