gnuradio/gnuradio #986: Avoid unnecessary memcpy() in some basic math operation blocks

mbr0wn, 2016-09-17:
A few basic math operation blocks that support a variable number of inputs (add_ff, multiply_ff, multiply_cc, etc.), always perform a usually unnecessary memcpy() to support the extreme case of only 1 input stream.

These blocks can get a performance boost for the very common case of 2 input streams by rewriting the loop to avoid the memcpy(), if the number of input items is greater than 1.

A revised add_ff work() body might look like this:

if (input_items.size() == 1) {
    memcpy(out, input_items[0], noi*sizeof(float));
} else {
      volk_32f_x2_add_32f(out, (const float *)input_items[0], (const float*)input_items[1], noi);
      for(size_t i = 2; i < input_items.size(); i++)
        volk_32f_x2_add_32f(out, out, (const float*)input_items[i], noi);
}
return noutput_items;

mbr0wn, 2016-09-17:
Author: Martin Braun
Date: Sat, 12 December 2015  2:35:06

- **Status** changed from New to Feedback
Such a patch should probably include branch hints, since 1 channel is unlikely.

	I'd be curious to see how much improvement this yields. It's a small fix, and even small improvements are probably worth the patch, but it's an interesting question.

mbr0wn, 2016-09-17:
Author: Andy Walls
Date: Sun, 13 December 2015 22:57:13

- **File** [add_perf_test.grc.png][1]  added - **File** [add_ff_100Msps_under_operf.txt][2] [![Magnifier][3] ][4]  added - **File** [add_nocopy_ff_100Msps_under_operf.txt][5] [![Magnifier][6] ][7]  added - **File** [pcpu_add_ff_100Msps_under_operf.txt][8] [![Magnifier][9] ][10]  added - **File** [pcpu_add_ff_100Msps_not_under_operf.txt][11] [![Magnifier][12] ][13]  added
Attaching add_ff and add_nocopy_ff comparison running at 100 Msps on an x86_64 system with a Core i7 processor.

	~4.5 %CPU difference when not running under operf.~3.3 %CPU difference running under operf.

	operf shows add_ff thread, when outside of kernel, is dominated by __memcpy_sse2_unaligned.operf shows add_nocopy_ff thread, when outside of kernel, is dominated by orcexec.TBZgbE (the ORC instantiation of the volk function to add 32-bit floats).

	Proving that the optimal way to perform a memcpy() is to figure out a way to not need it. :)

    

  [1]: /redmine/attachments/download/918/add_perf_test.grc.png
  [2]: /redmine/attachments/download/920/add_ff_100Msps_under_operf.txt
  [3]: /redmine/images/magnifier.png?1369394539
  [4]: /redmine/attachments/920/add_ff_100Msps_under_operf.txt
  [5]: /redmine/attachments/download/919/add_nocopy_ff_100Msps_under_operf.txt
  [6]: /redmine/images/magnifier.png?1369394539
  [7]: /redmine/attachments/919/add_nocopy_ff_100Msps_under_operf.txt
  [8]: /redmine/attachments/download/921/pcpu_add_ff_100Msps_under_operf.txt
  [9]: /redmine/images/magnifier.png?1369394539
  [10]: /redmine/attachments/921/pcpu_add_ff_100Msps_under_operf.txt
  [11]: /redmine/attachments/download/922/pcpu_add_ff_100Msps_not_under_operf.txt
  [12]: /redmine/images/magnifier.png?1369394539
  [13]: /redmine/attachments/922/pcpu_add_ff_100Msps_not_under_operf.txt

mbr0wn, 2016-09-17:
Author: Andy Walls
Date: Mon, 14 December 2015 15:16:51

- **File** [pcpu_add_ff_10Msps_not_under_operf.txt][1] [![Magnifier][2] ][3]  added
Adding a %CPU comparison measurement at 10 Msps that also include CPU time.

	Still about 1 %CPU difference at 10 Msps.

	At 5 Msps, I can't see a % CPU difference anymore, but over time, the accumulated CPU time differs.

    

  [1]: /redmine/attachments/download/923/pcpu_add_ff_10Msps_not_under_operf.txt
  [2]: /redmine/images/magnifier.png?1369394539
  [3]: /redmine/attachments/923/pcpu_add_ff_10Msps_not_under_operf.txt

mbr0wn, 2016-09-17:
Author: Martin Braun
Date: Mon, 21 December 2015  1:06:48

I'm convinced. Should make a pull request for this.

marcusmueller, 2020-10-30:
Hey, that's still not fixed in `add_blk_impl`, @mbr0wn! This is a really nice beginner issue, though.

mormj, 2022-08-24:
This appears to have been fixed in #3615 and #1968

closed 2022-08-24 by mormj