gnuradio/gnuradio #989: qa_fecapi_ldpc.py test when run in a 32bit system can hang indefinitely using 100% of one core

mbr0wn, 2016-09-17:
During rpm build on the Mageia Build System, qa_fecapi_ldpc.py hangs the system with one core (of 24) at 100% until that test is killed, after which all the other tests complete (with a few expected fails).
This only affects the 32bit build.
I reproduced this in a chroot locally (on a 4 core machine) with exactly the same outcome.
We have temporarily disabled that test in the 32bit build, but raising a bug here so it may be addressed properly.

mbr0wn, 2016-09-17:
Author: Barry Jackson
Date: Thu, 31 December 2015 16:01:34

test 87        Start  87: qa_fecapi_ldpc

	87: Test command: /usr/bin/sh "/home/iurt/rpmbuild/BUILD/gnuradio-3.7.9/build/gr-fec/python/fec/qa_fecapi_ldpc_test.sh" 87: Test timeout computed to be: 9.99988e+0687: /home/iurt/rpmbuild/BUILD/gnuradio-3.7.9/build/gr-fec/python/fec/qa_fecapi_ldpc_test.sh: line 9: 26551 Terminated              /usr/bin/python2 -B /home/iurt/rpmbuild/BUILD/gnuradio-3.7.9/gr-fec/python/fec/qa_fecapi_ldpc.py 87/207 Test  [#87][1] : qa_fecapi_ldpc .......................***Failed  15333.70 sec

    

  [1]: http://gnuradio.org/redmine/issues/87

mbr0wn, 2016-09-17:
Author: Barry Jackson
Date: Thu, 31 December 2015 17:03:55

Output as requested on irc (run from rpm build spec during build in chroot with just the one test). It hung at that point with 100% of one core in use.

	ctest -V -R ldpcUpdateCTestConfiguration  from :/home/baz/rpmbuild/BUILD/gnuradio-3.7.9/build/DartConfiguration.tclUpdateCTestConfiguration  from :/home/baz/rpmbuild/BUILD/gnuradio-3.7.9/build/DartConfiguration.tclTest project /home/baz/rpmbuild/BUILD/gnuradio-3.7.9/buildConstructing a list of testsDone constructing a list of testsChecking test dependency graph...Checking test dependency graph endtest 93    Start 93: qa_fecapi_ldpc

	93: Test command: /usr/bin/sh "/home/baz/rpmbuild/BUILD/gnuradio-3.7.9/build/gr-fec/python/fec/qa_fecapi_ldpc_test.sh" 93: Test timeout computed to be: 9.99988e+0693: /home/baz/.gnuradio/prefs/vmcircbuf_default_factory: No such file or directory93: vmcircbuf_createfilemapping: createfilemapping is not available

barjac, 2019-11-22:
Still happening in 3.8.0.0

argilo, 2022-10-15:
I recently ran into this. I think the hang happens because the `data_in` and `data_out` lists mismatch in this test:

https://github.com/gnuradio/gnuradio/blob/09930fdcc5c892eaff5754f6ef0203146bc1cca7/gr-fec/python/fec/qa_fecapi_ldpc.py#L108-L122

The lists are very long (~50k elements) and `assertEqual` is painfully slow in this circumstance.

I don't yet know why the mismatch occurs.

argilo, 2022-10-15:
I tracked down the bug; #6254 should fix it.

PR #6254 (2022-10-21): fec: Fix LDPC output size calculation
## Description
As noted in #989 and #5973, `qa_fecapi_ldpc` fails on i686 systems. This happens because the output size of the LDPC encoder is miscalculated. The frame size is multiplied by the code rate in floating point, and the result is cast to an integer without rounding. As a result, the output size may be one too small.

## Related Issue
Fixes #989.

## Which blocks/areas does this affect?
* LDPC Generator Matrix
* LDPC Parity Check Matrix

## Testing Done
I verified that the QA tests pass on i686 and x86_64.

## Checklist
- [x] I have read the [CON [... cut]
closed 2022-10-21 by mormj