gnuradio/gnuradio #991: frequent hang initializing dongle with macosx 10.11 El Capitan

mbr0wn, 2016-09-17:
problem is frequent failure on macosx 10.11 of gnuradio to complete
initialization with dongle (apparent hang with little CPU used)

environment is
 macosx 10.11.2 El Capitan, 2.4 GHz Intel Core 2 Duo, 4GB ram
 gnuradio 3.7.9_1+docs+grc+jack+portaudio+qtgui+sdl+swig+uhd+wavelet+wxgui
 using "Keedox RTL-SDR, FM+DAB, DVB-T USB Stick Set with RTL2832U & R820T"

Originally, this all was used successfully with macosx 10.9.5 Mavericks,
but after the upgrade(?) to El Capitan, reinstall of XCode and MacPorts,
and reinstall of the gnuradio bits, attempts to run previous things that
had worked with gnuradio-companion now frequently hang up in initialization.

GQRX (Gqrx 20151228) shows similar frequent but not uniform failure.

rtl_fm works every (20) time.

When successful, the terminal output includes the following lines:
  gr-osmosdr v0.1.x-xxx-xunknown (0.1.5git) gnuradio 3.7.9
  built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy 
  Using device #0 Realtek RTL2838UHIDIR SN: 00000001
  Found Rafael Micro R820T tuner
  ...

When hang happens, the "built-in" line prints, seemingly from
  gr-osmosdr/lib/source_impl.cc

but no output shows up after that.  The "Using device" line source seems in
  gr-osmosdr/lib/rtl/rtl_source_c.cc

Is this something others have experienced or should I begin digging deeper
with tools like "python -m trace" or dtruss?

mbr0wn, 2016-09-17:
Author: Michael Dickens
Date: Mon, 04 January 2016  0:17:31

This bug sounds most like a gr-osmosdr issue, rather than a GR issue per se. Can you send me an email ( [michael.dickens@ettus.com][1]  ) with some info in it &#38; I'll help you off-ticket? Here's a good starting point:
port installed and active
env
uname -a
and, if/when the crash happens see if you can track down a Console.app crash log &#38; send that along too.

    

  [1]: mailto:michael.dickens@ettus.com

mbr0wn, 2016-09-17:
Author: mike brescia
Date: Tue, 19 January 2016  4:49:18

Note: no crash happens, just a hang (well, there is a record after ForceQuit)

	What works and what doesn't with GRC, GnuRadio, gr-osmosdr in MacBookPro running El Capitan.

	Note that MacPorts is run to upgrade any outdated sw in the hope that bugs get fixed.

	1. a WX-GUI script will sometimes hang.  After a hang, attempts to retry also hang.

	2. a NO-GUI script does not hang.  Ever.  After a successful NO-GUI script, it has not healed anything because a following WX-GUI script will hang.

	3. after a while idle, (no GnuRadio activity) a WX-GUI script will succeed once or a few times, then fail.  Have not bracketed the time yet.

	4. running as root with a WX-GUI script succeeds, even after a run without root hung.

	5. turning off SystemIntegrityProtection (SIP) does NOT make anything work better.

mbr0wn, 2016-09-17:
Author: Martin Braun
Date: Fri, 22 January 2016 14:34:17

- **Status** changed from New to Feedback
Still not sure if this is even an GNU Radio bug.

mbr0wn, 2016-09-17:
Author: Michael Dickens
Date: Tue, 16 February 2016 20:23:21

- **Assignee** set to Michael Dickens - **Resolution** set to worksforme
After extensive debugging off-ticket, I believe that the issue is lots of cruft that was already installed & competing with the MacPorts install of GR / UHD / Volk. The OP has found a way around this issue for now, so I'm closing this ticket as worksforme.

mbr0wn, 2016-09-17:
@michaelld has this. Can't assign -- not yet in coredevs team.

mbr0wn, 2016-09-17:
@michaelld

michaelld, 2016-09-17:
@mbr0wn where did this come from?

mbr0wn, 2016-09-17:
I don't know. These issues are only authored by me because I ran the redmine import.

mbr0wn, 2016-09-17:
Might be BS. Feel free to close if it is. I haven't read most of these issues actually.

jmcorgan, 2016-09-17:
Now you own ALL THE THINGS

michaelld, 2016-10-06:
I see no reason to keep this issue open. I helped the person out; we came to a resolution, whatever it was. The OP had a workaround; have heard nothing back since. I'm not the assignee, so I can't close.

closed 2016-10-06 by jmcorgan