gnuradio/gnuradio #993: cmake enables support for ctrlport even when thrift is not found

mbr0wn, 2016-09-17:
Also, thrift is not listed in some of the installation guides, which leads me to believe few ever notice they don't actually have ctrlport

mbr0wn, 2016-09-17:
Author: Tom Rondeau
Date: Fri, 15 January 2016 15:00:00

The manual lists Thrift as a dependency:[http://gnuradio.org/doc/doxygen/build_guide.html][1] 

	But the installation guides should be updated. On the other hand, we've always been terrible about keeping our installation guides up to date with the number of distros supported and how fast things can change. One reason we've been pushing our installers like PyBOMBS more and more.

	As for ControlPort being enabled even without Thrift, that was done on purpose. ControlPort doesn't need Thrift or any backend to be compiled. It's just that it isn't really usable in that state, but everything will still compile and run. We did add the tracking of the backend (thrift only, so far) into the CMake structure, so we could potentially test if any backends are enabled, and if that number is == 0 then we disable ControlPort. I'm not opposed to that behavior if someone writes the appropriate CMake code to do so.

    

  [1]: http://gnuradio.org/doc/doxygen/build_guide.html

jmcorgan, 2016-09-18:
We make it possible to enable ctrlport, even when there is no transport found, so that code that compiles against libgnuradio-runtime will continue to compile.  The ctrlport functions are NOP in this case, so it is normal.

closed 2016-09-18 by jmcorgan

pinkavaj, 2016-09-18:
Should not be at leas displayed some warning after 'cmake'? (Eg in 'Gnuradio enabled components',)

jmcorgan, 2016-09-18:
It could warn "No transport found" when listing the transports under ctrlport in the enabled components.  Though, the CMake might be ugly for that.  Feel free to experiment :+1: