gnuradio/gnuradio #998: Error building gnuradio-3.7.9.1 in GCC-6

mbr0wn, 2016-09-17:
GCC-6 switched default C++ mode from GNU++98 to GNU++14/C++14. As a result you cannot have static float consts defined in classes. You need either:

- refactoring the code
- using -std=c++98 command line switch
- using 'constexpr' instead of 'const' in C++11 or newer modes

Attaching patch for the last mentioned solution utilizing BOOST defines, but feel free to use different approach.

mbr0wn, 2016-09-17:
Author: Johnathan Corgan
Date: Wed, 10 February 2016 18:20:41

- **Category** set to build - **Status** changed from New to Assigned - **Assignee** set to Johnathan Corgan
Thanks for bringing this to our attention.  Can you submit a pull request off the maint branch to our github?

mbr0wn, 2016-09-17:
Author: Jaroslav Škarvada
Date: Wed, 10 February 2016 18:25:47

Sure, but it needs some love. I will try to de-duplicate the code and provide e.g. _CONST_EXPR which will be conditionally set according to the standard used. Or maybe we could avoid this machinery completely and just avoid the static consts. I will take deeper look.

mbr0wn, 2016-09-17:
Author: Jaroslav Škarvada
Date: Fri, 12 February 2016 17:33:07

Pull request: [https://github.com/gnuradio/gnuradio/pull/749][1] 

	A bit better than the previously attached patch, but still nothing more than a quick hack.

    

  [1]: https://github.com/gnuradio/gnuradio/pull/749

mbr0wn, 2016-09-17:
Author: Jaroslav Škarvada
Date: Tue, 16 February 2016 17:05:17

I updated the github pull request. Hope it compiles now OK for you. Unfortunately it is probably not possible to change the pull request to be against 'maint' branch. Should I close it and open new one?

mbr0wn, 2016-09-17:
Author: Johnathan Corgan
Date: Tue, 16 February 2016 17:26:23

No worries, I'll cherry pick it.

jmcorgan, 2016-09-18:
This was fixed in #749.

closed 2016-09-18 by jmcorgan