gnuradio/gnuradio #1039: GNU Radio: Add "flow control" mechanism to response port #685

balister, 2016-09-28:
Martin made me do this.

jmcorgan, 2016-09-28:
@mbr0wn Can you be more descriptive about what this is?

mbr0wn, 2016-09-28:
This is about querying the buffer fullness of the device e.g. from a message port. For building low-latency apps, this is nice, because source blocks can be smarter about generating data.

jmcorgan, 2016-09-29:
Ah, got it.  I did something sort of like this once a while back by having a block near the end of a flowgraph send an async message back to a source block advertising its buffer space remaining, like in TCP.

douggeiger, 2016-09-29:
@mbr0wn: is this something different from the buffer state that is currently available via ctrlport?

mbr0wn, 2016-09-29:
Yes, this is about the buffer state on the hardware (not the block).

douggeiger, 2016-09-29:
Ah, understood, thanks!

marcusmueller, 2023-10-24:
I think this gotten solved through @dkozel & @MattEttus writing that flow control OOT

closed 2023-10-24 by marcusmueller