gnuradio/gnuradio #1040: Segmentation fault on PMT variable access

mowfask, 2016-09-29:
Apparently PMT variables may still be accessed in python after they have been deallocated in C++ land. This happens when the reference counting in C++ land hits zero, while python still holds references to the object.

A test case can be found [here](https://gist.github.com/mowfask/37ae4734da36f1439d138a918653785c), @marcusmueller started digging into this [in this email](https://lists.gnu.org/archive/html/discuss-gnuradio/2016-07/msg00152.html).

My current understanding is that swig needs to be told about this situation explicitly in https://github.com/gnuradio/gnuradio/blob/master/gnuradio-runtime/swig/pmt_swig.i, e.g. using the swig "ref" and "unref" features (see [swig documentation](http://www.swig.org/Doc1.3/Python.html#Python_nn27a)).

Tested against the current master (9e11c519b55d65bfcedfa5324c75bf7f268c9dbc) and previously confirmed by Marcus (see email).

jdeitmerg, 2020-01-27:
Sorry, I didn't mean to unassign anyone. I just updated the link to the test case as the old one was dead.

marcusmueller, 2020-10-30:
Is this still the case after Pybindization?

mormj, 2020-10-30:
Yes this appears to be OBE on master - cannot reproduce, likely due to pybind11

mormj, 2021-03-24:
Closing this since the issue can't be reproduced after pybind11 replaced swig

closed 2021-03-24 by mormj