gnuradio/gnuradio #1044: Linking CXX shared library libgnuradio-core-3.6.4.2.so

stackprogramer, 2016-10-01:
when i try to make gnuradio 3.6.4.2 it returns errors........
i build gnuradio 3.7 with out problems.but for airprobe i need this version. how can fixed this error????????

> Linking CXX shared library libgnuradio-core-3.6.4.2.so [ 33%] Built target gnuradio-core Scanning dependencies of target gnuradio-config-info [ 33%] Building CXX object gnuradio-core/src/lib/CMakeFiles/gnuradio-config-info.dir/gnuradio-config-info.cc.o Linking CXX executable gnuradio-config-info [ 33%] Built target gnuradio-config-info Scanning dependencies of target gr_core_rstest [ 33%] Building C object gnuradio-core/src/lib/CMakeFiles/gr_core_rstest.dir/reed-solomon/encode_rs.c.o [ 33%] Building C object gnuradio-core/src/lib/CMakeFiles/gr_core_rstest.dir/reed-solomon/decode_rs.c.o [ 33%] Building C object gnuradio-core/src/lib/CMakeFiles/gr_core_rstest.dir/reed-solomon/init_rs.c.o [ 33%] Building C object gnuradio-core/src/lib/CMakeFiles/gr_core_rstest.dir/reed-solomon/rstest.c.o [ 33%] Building C object gnuradio-core/src/lib/CMakeFiles/gr_core_rstest.dir/reed-solomon/exercise.c.o Linking C executable gr_core_rstest

mbr0wn, 2016-10-01:
Please post support issues on the mailing list, not on the bug tracker.
However I'm not sure how much help you'll get for an ancient version of GNU
radio. There's better tools than airprobe now.

stackprogramer, 2016-10-02:
so i used gnuradio-3.3.0 ,it return errors 

> configure: error: Could not link against boost_thread
i fixed it with 
`  ./configure --with-boost-libdir=/usr/lib/x86_64-linux-gnu
`
now it returns 

> errors:io-core/src/lib/libgnuradio-core.la' or unhandled argument `/home/sp/Documents/gnuradio-3.3.0/gnuradio-core/src/lib/libgnuradio-core.la'
> Makefile:1072: recipe for target 'libgnuradio-qtgui.la' failed
> make[4]: *** [libgnuradio-qtgui.la] Error 1
> make[4]: Leaving directory '/home/sp/Documents/gnuradio-3.3.0/gr-qtgui/src/lib'
> Makefile:1299: recipe for target 'check' failed
> make[3]: *** [check] Error 2
> make[3]: Leaving directory '/home/sp/Documents/gnuradio-3.3.0/gr-qtgui/src/lib'
> Makefile:672: recipe for target 'check-recursive' failed
> make[2]: *** [check-recursive] Error 1
> make[2]: Leaving directory '/home/sp/Documents/gnuradio-3.3.0/gr-qtgui/src'
> Makefile:780: recipe for target 'check-recursive' failed
> make[1]: *** [check-recursive] Error 1
> make[1]: Leaving directory '/home/sp/Documents/gnuradio-3.3.0/gr-q

tgui'
Makefile:837: recipe for target 'check-recursive' failed
make: *** [check-recursive] Error 1

jmcorgan, 2016-10-02:
Again, please ask questions like these on the GNU Radio mailing list.

closed 2016-10-02 by jmcorgan