gnuradio/gnuradio #1045: log4cpp-related linking error with OOT-packages when GNURadio was built with log4cpp

mnhauke, 2016-10-02:
I've just built GNURadio 3.7.10.1 with log4cpp support but now gr-osmosdr fails to compile because of "undefined reference to log4cpp:Appender..."  errors since it does not have the information to also link against log4cpp.

Here's the full buildlog from gr-osmosdr:
https://blackhole.pmtu.de/.dump/gr-osmosdr-buildlog-fail.txt

I actually worked around by adding log4cpp to the linker flags manually but it still does not feel like the proper way to address this problem...
https://blackhole.pmtu.de/.dump/gr-osmosdr-fix-log4cpp-linking-error.diff

Conditionally adding the lib4cpp linking flags to gnuradio-runtime.pc could be a solution.

GNUradio was build in a clean chroot-environment.
CMake automatically detected log4cpp and set ENABLE_GR_LOG to ON.

Here's a the GNURadio buildlog:
https://blackhole.pmtu.de/.dump/gnuradio-buildlog-log4cpp.txt

libgnuradio-runtime.so is linked to liblog4cpp.so and so should all programs that make use of this library ...
https://blackhole.pmtu.de/.dump/ldd_libgnuradio-runtime_so.txt

bastibl, 2017-01-06:
I think this is just one of many symptoms of GNU Radio's lack of dependency tracking in cmake.

On my mac, the problem already starts with the includes. Since ${GNURADIO_RUNTIME_INCLUDE_DIRS} is just another name for <install_prefix>/include, it already fails when looking for log4cpp headers. (On Linux that's often not an issue since the headers are in a standard directory that's likely pulled in by another library. Every module uses boost; so if the headers of boost and log4cpp are in the same directory, it works by chance.)
The problem is that every block includes `block.h`, which includes `logger.h`, which includes `log4cpp`. So every module that uses gnuradio-runtime has to know about its dependency on log4cpp and manually add the includes.

Sticking to the current definition of ${GNURADIO_RUNTIME_INCLUDE_DIRS}, the problem already starts in-tree. To compile GNU Radio on my mac, I had to [add log4cpp in every module](https://github.com/bastibl/gnuradio/commit/16d8e8cb9bd658991dabbc253c74850248b80f83).
That's clearly not so nice. I think that, ${GNURADIO_RUNTIME_INCLUDE_DIRS} should include everything that is needed when *using* gnuradio-runtime, i.e., all headers that are referenced by its public headers.

The same problem is also apparent for the libraries. I think, ${GNURADIO_ALL_LIBRARY_DIRS} should include the directories of the shared libraries that it uses. Currently, it only includes the directories of libgnuradio-xxx, which leads to the above mentioned problem.

Even in-tree, there seems to be lots of confusion on how to deal with shared libraries that are indirectly referenced through other modules. Looking again at log4cpp:
- Some modules just link against runtime and that’s it (see gr-analog).
- Some modules put ${LOG4CPP_LIBRARY_DIRS} in the search path and link against runtime (see gr-trellis).
- Some modules put ${LOG4CPP_LIBRARY_DIRS} in the search path and link against runtime and log4cpp even though they don’t use it directly (see gr-audio).
- Some modules link against runtime and log4cpp even though they don’t use it directly (see gr-blocks).

Note for in-tree modules the easiest solution, i.e. to just link against runtime, as gr-analog does, is OK. During compilation cmake sets the rpath (or uses the install name on macOS) and, therefore, finds libraries during compilation. The rpath is, however, stripped during installation, which is why OOT modules might not be able to find the libraries (as shown by the above build logs).

To deal with this, we could consider to:
- redefine ${GR_XXX_INCLUDE_DIRS} to include all directories referenced by its public headers
- introduce ${GR_XXX_LIBRARY_DIRS} to include the directories of all referenced shared libraries.
- store these values in GnuradioConfig.cmake to provide them to OOT modules

It would be interesting to hear your thoughts on that.

jmcorgan, 2017-01-26:
@bastibl I had missed this when you added your thoughtful comments above. I think you're on the right track, but we'd need to figure out how to actually automate generating the above CMake variables.

vamsi765, 2017-04-24:
Hi,

Is this resolved? I still face the issue with log4cpp on Mac, where I installed gnuradio using Macport developer version.

> gnuradio-config-info -v
v3.7-MacPorts-devel-git-3a2c67c6(20170413)

https://github.com/gnuradio/gnuradio/commit/3a2c67c67639e0a43ec90e69199c0e4b0cf1b05b

Issue:
[ 17%] Linking CXX shared library libgnuradio-fsk_mod.dylib
Undefined symbols for architecture x86_64:
  "log4cpp::Appender::AppenderMapStorageInitializer::AppenderMapStorageInitializer()", referenced from:
      __GLOBAL__sub_I_fsk_mod_impl.cc in fsk_mod_impl.cc.o
      __GLOBAL__sub_I_stream_gatepass_impl.cc in stream_gatepass_impl.cc.o
      __GLOBAL__sub_I_threshold_detect_impl.cc in threshold_detect_impl.cc.o
  "log4cpp::Appender::AppenderMapStorageInitializer::~AppenderMapStorageInitializer()", referenced from:
      __GLOBAL__sub_I_fsk_mod_impl.cc in fsk_mod_impl.cc.o
      __GLOBAL__sub_I_stream_gatepass_impl.cc in stream_gatepass_impl.cc.o
      __GLOBAL__sub_I_threshold_detect_impl.cc in threshold_detect_impl.cc.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [lib/libgnuradio-fsk_mod.dylib] Error 1
make[1]: *** [lib/CMakeFiles/gnuradio-fsk_mod.dir/all] Error 2
make: *** [all] Error 2

Note: I have this code which worked two years ago[version 3.7 I think].

Is there a temporary workaround for this for immediate use? I have tried the fix in below issue with no success:
https://github.com/gnuradio/gr-tutorial/issues/10

Thanks
Vamsi

bastibl, 2017-04-24:
So you are trying to build an OOT module against GNU Radio master? Is the code somewhere online?

I don't use MacPorts. Does it compile GNU Radio with log4cpp support?
Do you still have log4cpp installed? 

Can you run `otool -L libgnuradio-runtime.dylib` (might be somewhere under `/opt/`)

What libraries do you link your module against? Did you try the fix suggested in this thread?

Can you post the linker error after `VERBOSE=1 make`?

vamsi765, 2017-04-25:
> So you are trying to build an OOT module against GNU Radio master?

 Almost, As I understand Macport gnuradio devel version updates every week or so.

> Is the code somewhere online?

Any code works the same, simple module with a block, I have put a simple code on GitHub
https://github.com/vamsi765/gr-testinggnuradio

There is another uploaded tutorial facing the same issue.
https://github.com/gnuradio/gr-tutorial/issues/10#issuecomment-296791088

> I don't use MacPorts. Does it compile GNU Radio with log4cpp support?
Do you still have log4cpp installed?

Yes Macport has the dependency: Log4cpp is available [There is no compile time issue, just during linking]

>Can you run otool -L libgnuradio-runtime.dylib (might be somewhere under /opt/)

  otool -L /opt/local/lib/libgnuradio-runtime.dylib 
/opt/local/lib/libgnuradio-runtime.dylib:
	/opt/local/lib/libgnuradio-runtime.3.7.12git.dylib (compatibility version 3.7.12, current version 0.0.0)
	/opt/local/lib/libgnuradio-pmt.3.7.12git.dylib (compatibility version 3.7.12, current version 0.0.0)
	/opt/local/lib/libvolk.1.3.dylib (compatibility version 1.3.0, current version 0.0.0)
	/opt/local/lib/libboost_date_time-mt.dylib (compatibility version 0.0.0, current version 0.0.0)
	/opt/local/lib/libboost_program_options-mt.dylib (compatibility version 0.0.0, current version 0.0.0)
	/opt/local/lib/libboost_filesystem-mt.dylib (compatibility version 0.0.0, current version 0.0.0)
	/opt/local/lib/libboost_system-mt.dylib (compatibility version 0.0.0, current version 0.0.0)
	/opt/local/lib/libboost_regex-mt.dylib (compatibility version 0.0.0, current version 0.0.0)
	/opt/local/lib/libboost_thread-mt.dylib (compatibility version 0.0.0, current version 0.0.0)
	/opt/local/lib/libboost_chrono-mt.dylib (compatibility version 0.0.0, current version 0.0.0)
	/opt/local/lib/libboost_atomic-mt.dylib (compatibility version 0.0.0, current version 0.0.0)
	/opt/local/lib/liblog4cpp.5.dylib (compatibility version 6.0.0, current version 6.6.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 307.5.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1238.51.1)

> What libraries do you link your module against? Did you try the fix suggested in this thread?

I'm not sure what the fix is? Adding log4cpp dependency? if so, I'm not sure how to do that.

> Can you post the linker error after VERBOSE=1 make?

[  4%] Linking CXX shared library libgnuradio-testinggnuradio.dylib
cd /Users/vamsi/Documents/workspace/gr-testinggnuradio/build/lib && /opt/local/bin/cmake -E cmake_link_script CMakeFiles/gnuradio-testinggnuradio.dir/link.txt --verbose=1
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang -O3 -DNDEBUG -dynamiclib -Wl,-headerpad_max_install_names -compatibility_version 1.0.0 -o libgnuradio-testinggnuradio.1.0.0git.dylib -install_name /opt/local/lib/libgnuradio-testinggnuradio.1.0.0git.dylib CMakeFiles/gnuradio-testinggnuradio.dir/block1_impl.cc.o  -L/opt/local/lib -Wl,-rpath,/opt/local/lib /opt/local/lib/libboost_filesystem-mt.dylib /opt/local/lib/libboost_system-mt.dylib /opt/local/lib/libgnuradio-runtime.dylib /opt/local/lib/libgnuradio-pmt.dylib 
Undefined symbols for architecture x86_64:
  "log4cpp::Appender::AppenderMapStorageInitializer::AppenderMapStorageInitializer()", referenced from:
      __GLOBAL__sub_I_block1_impl.cc in block1_impl.cc.o
  "log4cpp::Appender::AppenderMapStorageInitializer::~AppenderMapStorageInitializer()", referenced from:
      __GLOBAL__sub_I_block1_impl.cc in block1_impl.cc.o
  "std::bad_exception::what() const", referenced from:
      vtable for boost::exception_detail::bad_exception_ in block1_impl.cc.o
      vtable for boost::exception_detail::clone_impl<boost::exception_detail::bad_exception_> in block1_impl.cc.o
  "std::runtime_error::what() const", referenced from:
      vtable for boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::bad_function_call> > in block1_impl.cc.o
      vtable for boost::exception_detail::error_info_injector<boost::bad_function_call> in block1_impl.cc.o
      vtable for boost::bad_function_call in block1_impl.cc.o
  "std::__1::__vector_base_common<true>::__throw_length_error() const", referenced from:
      gr::block::processor_affinity() in block1_impl.cc.o
  "std::bad_alloc::what() const", referenced from:
      vtable for boost::exception_detail::bad_alloc_ in block1_impl.cc.o
      vtable for boost::exception_detail::clone_impl<boost::exception_detail::bad_alloc_> in block1_impl.cc.o
  "std::bad_exception::~bad_exception()", referenced from:
      boost::exception_ptr boost::exception_detail::get_static_exception_object<boost::exception_detail::bad_exception_>() in block1_impl.cc.o
      boost::exception_detail::clone_impl<boost::exception_detail::bad_exception_>::clone_impl(boost::exception_detail::bad_exception_ const&) in block1_impl.cc.o
      boost::exception_detail::clone_impl<boost::exception_detail::bad_exception_>::~clone_impl() in block1_impl.cc.o
      boost::exception_detail::bad_exception_::~bad_exception_() in block1_impl.cc.o
      boost::exception_detail::bad_exception_::~bad_exception_() in block1_impl.cc.o
      non-virtual thunk to boost::exception_detail::bad_exception_::~bad_exception_() in block1_impl.cc.o
      non-virtual thunk to boost::exception_detail::bad_exception_::~bad_exception_() in block1_impl.cc.o
[... 117 lines cut ...]
[... cut ...]
      virtual thunk to gr::testinggnuradio::block1::~block1() in block1_impl.cc.o
      gr::basic_block::dispatch_msg(boost::intrusive_ptr<pmt::pmt_base>, boost::intrusive_ptr<pmt::pmt_base>) in block1_impl.cc.o
      ...
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [lib/libgnuradio-testinggnuradio.1.0.0git.dylib] Error 1
make[1]: *** [lib/CMakeFiles/gnuradio-testinggnuradio.dir/all] Error 2
make: *** [all] Error 2

vamsi765, 2017-04-26:
I was able to fix this issue [Carles also has similar changes required]:
This commit shows what the changes to be made for a simple OOT block to work
https://github.com/vamsi765/gr-testinggnuradio/commit/c204fc73a1d040539298f776d2fcd345fcce6394

bastibl, 2017-04-27:
Great that it fixes your problem, but think it's a workaround rather than a real solution. Without further support from GNU Radio, the OOT would have to test if GNU Radio was build with log4cpp and (in case) search for the library, make sure that it's the same version, and link against it.
That's quite some complexity that would have to be added to all OOT modules. Considering, for example, runtime, I think it would be better if `GNURADIO_RUNTIME_INCLUDE_DIRS` returned all include directories that are needed when using runtime (i.e.,  all headers that are reachable through the public headers in `include`).
Similarly, `GNURADIO_RUNTIME_LIBRARIES` should return all libraries that are required when linking against runtime.
It should be pretty easy to record these variables during build and would simplify OOT modules a lot. In particular,
- no need for GNU Radio configuration checks (like, is log4cpp used or not)
- GNU Radio and the OOT would always link to the same version of the library (this was often a problem for me on macOS, where I have often more than one version of a library installed)
- It doesn't require to update all OOTs when GNU Radio uses a new library.

Would be great to hear some thoughts on that.

(There is one thing I don't understand (and that's why I asked for all the command outputs): Why fail OOT builds with indirect shared library dependencies on mac  (in the scenario libgnuradio-tutorial.dylib -> libgnuradio-runtime -> liblog4cpp.dylib, the log4cpp symbols cannot be resolved, which leads to the above error.) But in-tree works (i.e., libgnuradio-blocks.dylib -> libgnuradio-runtime.dylib -> liblog4cpp.dylib) I know that during build shared objects use special settings for the rpath, for example, but I couldn't find something that impacts how symbols are resolved).)

jmcorgan, 2017-04-27:
Just as an aside, for GNU Radio 3.8+ and already on the next branch, logging is always turned on and log4cpp is a mandatory requirement, so this seems strictly a 3.7 issue.

bastibl, 2017-04-27:
I would say log4cpp is just one example where most problems occur. The suggestions were meant in general. On macOS, for example, I have the impression that 90% of the problems are because users link against the wrong version of a library (like system python vs. homebrew/ports python).
If find_package(gnuradio) would also return the proper dependencies that might simplify OOTs a lot.

jmcorgan, 2017-04-27:
I agree.  I'm not familiar with the Mac side of things--@michaelld looks after this--my comment was only an aside that the log4cpp complexity has been removed in 3.8.

michaelld, 2017-04-27:
The issue is with log4cpp 1.1.2. I'm still debugging. Revert to 1.1.1 if you can & it'll work nice & clean.

michaelld, 2017-04-27:
Many prior issues on macOS were because of the mixed Python issue as @bastibl points out. We've committed code that addresses that issue. This isn't that issue or even related to it. It's an issue with log4cpp & AppenderMapStorageInitializer. I haven't had time to debug it far enough to have a clue what's going on yet.

michaelld, 2017-04-27:
OK so the problem is the "static" declaration in the Appender class. The variable isn't used for anything anyway, and it's easy to get rid of: `sed -e "/appenderMapStorageInitializer/d"` on the file `include/log4cpp/Appender.hh`. I'm pushing this into MacPorts shortly.

michaelld, 2017-04-27:
OK. Fixed here in MacPorts: https://trac.macports.org/changeset/91dd35333bbcbd39eeaabf17ba037eeed20d9304/macports-ports .

vamsi765, 2017-04-28:
@michaelld Thanks for the update! That fixed the issue.

bastibl, 2017-04-28:
Great that it works, but a bit unfortunate that it works good enough to ignore the general problem with configuring OOTs.

I wanted to try the patch on homebrew and installed a fixed version of log4cpp somewhere in my home.  While I can point GR to use this version, I don't see how I can point OOTs to that location (if we don't add find(log4cpp) to all OOT modules). Since runtime references log4cpp in its include files, but doesn't add the log4cpp path to  `GNURADIO_ALL_INCLUDE_DIRS`, I cannot see how an OOT could find the headers.

So, AFAIS, with this slightly unusual installation, I cannot use (most) OOTs.

jmcorgan, 2017-04-28:
@bastibl I agree the more general issue is important to address.  Can you open a new issue so we can more properly address it outside this particular one?

closed 2017-04-28 by jmcorgan

michaelld, 2017-04-28:
It would be good to add the LOG4CPP_INCLUDE_DIRS as @bastibl did in his commit from his above comment, assuming that LOG4CPP is enabled. Might be enabled by default by now I don't recall.

michaelld, 2017-05-02:
After some searching I think the issue is with gnuradio-runtime/include/logger.h.in:38-43
{{{
#ifndef ENABLE_GR_LOG
#cmakedefine ENABLE_GR_LOG
#endif
#ifndef HAVE_LOG4CPP
#cmakedefine HAVE_LOG4CPP
#endif
}}}
where the CMake macro "#cmakedefine" replaces the line with either "#define ..." or "/* #undef ... */".

Let's look at the various possible use cases, for GR and an OOT, assuming that both ENABLE_GR_LOG and HAVE_LOG4CPP are both either enabled or disabled:

GR: True, OOT: True -> works
=> Both GR and OOT define ENABLE_GR_LOG and HAVE_LOG4CPP, so the above #define's aren't used.

GR: False, OOT: False -> works
=> Both GR and OOT do not define ENABLE_GR_LOG and HAVE_LOG4CPP, so the above #define's make no difference.

GR: False, OOT: True -> fails
=> GR is not compiled with support for logging, so OOTs can't use it either, no matter if requested or not.

GR: True, OOT: False -> fails
=> Since logging is either not enabled or explicitly turned off in the OOT, #include'ing logger.h should not force the OOT to link to log4cpp. But, in this case logging is actually enabled by these #define's & thus log4cpp is required.

Result: We need a better way to handle logging in OOTs & checking for when GR provides it & when it's not provided. I'll look into what it'll take to fix this & report back here.

michaelld, 2017-05-02:
In initial testing, the following works for me, replacing the above code. It's more complicated, but it handles the cases well. YA question is whether we also need to replace to code in the top-level config.in.h for these same macros ...

{{{
// handle current status of GR_LOG
#ifdef ENABLE_GR_LOG
#define EXT_ENABLE_GR_LOG
#undef ENABLE_GR_LOG
#endif

// did GR enable LOG?
#cmakedefine ENABLE_GR_LOG

// if GR does not provide logging and the current module is requesting
// it, disable it.
#if defined(EXT_ENABLE_GR_LOG) && !defined(ENABLE_GR_LOG)
#warning "Logging was requested but is not enabled in GNU Radio, so disabling."
#undef EXT_ENABLE_GR_LOG
#endif

// if GR provides logging but the current module is not requesting it,
// disable it.
#if !defined(EXT_ENABLE_GR_LOG) && defined(ENABLE_GR_LOG)
#warning "Logging was requested but is not enabled in GNU Radio, so disabling."
#undef ENABLE_GR_LOG
#endif

// the other 2 cases work; no need to check them!

// handle current status of LOG4CPP
#ifdef HAVE_LOG4CPP
#define EXT_HAVE_LOG4CPP
#undef HAVE_LOG4CPP
#endif

// did GR use log4cpp?
#cmakedefine HAVE_LOG4CPP

// if GR does not use log4cpp and the current module is requesting it,
// disable it & print a warning.
#if defined(EXT_HAVE_LOG4CPP) && !defined(HAVE_LOG4CPP)
#warning "Log4Cpp use was requested but was not in GNU Radio, so disabling."
#undef EXT_HAVE_LOG4CPP
#endif

// if GR provides for using log4cpp but the current module is not
// requesting it, disable it quietly.
#if !defined(EXT_HAVE_LOG4CPP) && defined(HAVE_LOG4CPP)
#undef HAVE_LOG4CPP
#endif

// the other 2 cases work; no need to check them!
}}}

jmcorgan, 2017-05-02:
And now you know firsthand why all this was ripped out on next for 3.8, with logging always enabled and log4cpp always required.

michaelld, 2017-05-02:
Yup! Is there a specific branch of next with the log4cpp changes, or is it on master?

jmcorgan, 2017-05-02:
It is set that way on the 'next' branch currently.

michaelld, 2017-05-02:
Yes of course not on master duh. OK, looking at that branch I see what you mean about making log4cpp required.

So, yes, then if not already done so we need to fix the GR OOT CMake scripts to always require log4cpp and include it for headers and library automagically, now that 1.1.2 is released.

This was not required with log4cpp 1.1.0: the header just declared the API such that an OOT that didn't explicitly use logging didn't require linking to log4cpp.

Version 1.1.2 added in a static variable that's instantiated just by including the header, and thus requires the library for linking even if the OOT does not explicitly use logging.

jmcorgan, 2017-05-02:
Good catch. I don't think the OOT module in gr-utils has been modified.

michaelld, 2017-05-02:
My preference would actually be to patch log4cpp to move the static variable to the .cpp file to make it benign again; push the fix upstream. This change would be because it's difficult to know when a header might indirectly include log4cpp.

The fix I mention would work for GR & should be compatible with most GR OOT modules. But it probably won't work with all of them & certainly not with non-GR projects.

My opinion: bad log4cpp for adding in that static variable in the header.

jmcorgan, 2017-05-02:
Of course, it's ideal if upstream will fix things.  And yes, a pox on anyone that puts things in header files that either allocates storage or forces linkage.

michaelld, 2017-05-04:
I think moving the static variable from the header to the .cpp code will do the trick. It does allow GNU Radio to execute again, which is a plus. I'm really not sure the intent of placing that static variable in the header; seems like there need be just 1 of them for the code to work correctly. Whatever. The commit is here: https://github.com/macports/macports-ports/commit/03e977add031301f57ca69729d829c607fe8d316 . I'll also create a ticket on the log4cpp SF area & link in that commit and this issue. Fun times!

michaelld, 2017-05-04:
https://sourceforge.net/p/log4cpp/support-requests/42/

pianoman730, 2019-05-29:
@michaelld Do we know if this issue has been fixed? I just ran into this using log4cpp 1.1.1 and the latest gnuradio. The error I ran into while running `make` is this:
```
[ 14%] Linking CXX executable pmt_unv
libgnuradio-runtime.so.3.8tech-preview-373-g8a7214c7: undefined reference to `log4cpp::Category::exists(std::string const&)'
libgnuradio-runtime.so.3.8tech-preview-373-g8a7214c7: undefined reference to `log4cpp::CategoryStream::operator<<(log4cpp::CategoryStream& (*)(log4cpp::CategoryStream&))'
libgnuradio-runtime.so.3.8tech-preview-373-g8a7214c7: undefined reference to `log4cpp::CategoryStream::~CategoryStream()'
libgnuradio-runtime.so.3.8tech-preview-373-g8a7214c7: undefined reference to `log4cpp::FileAppender::FileAppender(std::string const&, std::string const&, bool, unsigned int)'
libgnuradio-runtime.so.3.8tech-preview-373-g8a7214c7: undefined reference to `log4cpp::PropertyConfigurator::configure(std::string const&)'
libgnuradio-runtime.so.3.8tech-preview-373-g8a7214c7: undefined reference to `log4cpp::PatternLayout::PatternLayout()'
libgnuradio-runtime.so.3.8tech-preview-373-g8a7214c7: undefined reference to `log4cpp::Category::getInstance(std::string const&)'
libgnuradio-runtime.so.3.8tech-preview-373-g8a7214c7: undefined reference to `log4cpp::RollingFileAppender::RollingFileAppender(std::string const&, std::string const&, unsigned long, unsigned int, bool, unsigned int)'
libgnuradio-runtime.so.3.8tech-preview-373-g8a7214c7: undefined reference to `log4cpp::eol(log4cpp::CategoryStream&)'
libgnuradio-runtime.so.3.8tech-preview-373-g8a7214c7: undefined reference to `log4cpp::CategoryStream::operator<<(char const*)'
libgnuradio-runtime.so.3.8tech-preview-373-g8a7214c7: undefined reference to `log4cpp::OstreamAppender::OstreamAppender(std::string const&, std::ostream*)'
libgnuradio-runtime.so.3.8tech-preview-373-g8a7214c7: undefined reference to `log4cpp::Category::getCurrentCategories()'
collect2: error: ld returned 1 exit status
make[2]: *** [gnuradio-runtime/lib/pmt_unv] Error 1
make[1]: *** [gnuradio-runtime/lib/CMakeFiles/pmt_unv.dir/all] Error 2
make: *** [all] Error 2
```

michaelld, 2019-05-29:
I'm using stock `log4cpp` version 1.1.3 and can build the current GR GIT master HEAD without issue.

Can you do `make VERBOSE=ON` to get the actual link command? That might provide useful.

pianoman730, 2019-05-29:
Here's the make command on verbose
```
[ 14%] Linking CXX executable pmt_unv
cd /home/jacobhadley/gnuradio/build/gnuradio-runtime/lib && /usr/local/bin/cmake -E cmake_link_script CMakeFiles/pmt_unv.dir/link.txt --verbose=ON
/bin/c++   -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -O3 -DNDEBUG   CMakeFiles/pmt_unv.dir/pmt/qa_pmt_unv.cc.o  -o pmt_unv -Wl,-rpath,/home/jacobhadley/gnuradio/build/gnuradio-runtime/lib:/home/jacobhadley/gnuradio/build/gnuradio-runtime/lib/pmt:/home/jacobhadley/gnuradio/build/volk/lib libgnuradio-runtime.so.3.8tech-preview-381-g27dd99e4 pmt/libgnuradio-pmt.so.3.8tech-preview-381-g27dd99e4 /usr/lib64/libboost_unit_test_framework-mt.so ../../volk/lib/libvolk.so.1.4.1git -ldl /usr/lib64/liborc-0.4.so -lm /usr/lib64/libboost_program_options-mt.so /usr/lib64/libboost_filesystem-mt.so /usr/lib64/libboost_regex-mt.so /usr/lib64/libboost_thread-mt.so /usr/lib64/libboost_atomic-mt.so /usr/lib64/libboost_thread-mt.so /usr/lib64/libboost_atomic-mt.so -lpthread /usr/lib64/libboost_chrono-mt.so /usr/lib64/libboost_system-mt.so /usr/lib64/libboost_date_time-mt.so /usr/lib64/libgmpxx.so /usr/lib64/libgmp.so -lrt 
libgnuradio-runtime.so.3.8tech-preview-381-g27dd99e4: undefined reference to `log4cpp::Category::exists(std::string const&)'
libgnuradio-runtime.so.3.8tech-preview-381-g27dd99e4: undefined reference to `log4cpp::CategoryStream::operator<<(log4cpp::CategoryStream& (*)(log4cpp::CategoryStream&))'
libgnuradio-runtime.so.3.8tech-preview-381-g27dd99e4: undefined reference to `log4cpp::CategoryStream::~CategoryStream()'
libgnuradio-runtime.so.3.8tech-preview-381-g27dd99e4: undefined reference to `log4cpp::FileAppender::FileAppender(std::string const&, std::string const&, bool, unsigned int)'
libgnuradio-runtime.so.3.8tech-preview-381-g27dd99e4: undefined reference to `log4cpp::PropertyConfigurator::configure(std::string const&)'
libgnuradio-runtime.so.3.8tech-preview-381-g27dd99e4: undefined reference to `log4cpp::PatternLayout::PatternLayout()'
libgnuradio-runtime.so.3.8tech-preview-381-g27dd99e4: undefined reference to `log4cpp::Category::getInstance(std::string const&)'
libgnuradio-runtime.so.3.8tech-preview-381-g27dd99e4: undefined reference to `log4cpp::RollingFileAppender::RollingFileAppender(std::string const&, std::string const&, unsigned long, unsigned int, bool, unsigned int)'
libgnuradio-runtime.so.3.8tech-preview-381-g27dd99e4: undefined reference to `log4cpp::eol(log4cpp::CategoryStream&)'
libgnuradio-runtime.so.3.8tech-preview-381-g27dd99e4: undefined reference to `log4cpp::CategoryStream::operator<<(char const*)'
libgnuradio-runtime.so.3.8tech-preview-381-g27dd99e4: undefined reference to `log4cpp::OstreamAppender::OstreamAppender(std::string const&, std::ostream*)'
libgnuradio-runtime.so.3.8tech-preview-381-g27dd99e4: undefined reference to `log4cpp::Category::getCurrentCategories()'
collect2: error: ld returned 1 exit status
make[2]: *** [gnuradio-runtime/lib/pmt_unv] Error 1
make[2]: Leaving directory `/home/jacobhadley/gnuradio/build'
make[1]: *** [gnuradio-runtime/lib/CMakeFiles/pmt_unv.dir/all] Error 2
make[1]: Leaving directory `/home/jacobhadley/gnuradio/build'
make: *** [all] Error 2
```

pianoman730, 2019-05-29:
If it's helpful, here's the output from `Building CXX object gnuradio-runtime/lib/CMakeFiles/pmt_unv.dir/pmt/qa_pmt_unv.cc.o`:

```
[ 13%] Building CXX object gnuradio-runtime/lib/CMakeFiles/pmt_unv.dir/pmt/qa_pmt_unv.cc.o
cd /home/<user>/gnuradio/build/gnuradio-runtime/lib && /bin/c++  -DALIGNED_MALLOC=0 -DBOOST_TEST_DYN_LINK -DBOOST_TEST_MAIN -DGR_CTRLPORT -DGR_MPLIB_GMP -DGR_PERFORMANCE_COUNTERS -DHAVE_ARPA_INET_H -DHAVE_COSF -DHAVE_GETPAGESIZE -DHAVE_MALLOC_H -DHAVE_MMAP -DHAVE_NETINET_IN_H -DHAVE_PTHREAD_SETSCHEDPARAM -DHAVE_PTHREAD_SIGMASK -DHAVE_SCHED_SETSCHEDULER -DHAVE_SHM_OPEN -DHAVE_SIGACTION -DHAVE_SIGNAL_H -DHAVE_SINCOS -DHAVE_SINCOSF -DHAVE_SINF -DHAVE_SNPRINTF -DHAVE_SYSCONF -DHAVE_SYS_IPC_H -DHAVE_SYS_MMAN_H -DHAVE_SYS_RESOURCE_H -DHAVE_SYS_SHM_H -DHAVE_SYS_SOCKET_H -DHAVE_SYS_TIME_H -DHAVE_SYS_TYPES_H -DHAVE_UNISTD_H -DHAVE_VALLOC -I/home/<user>/gnuradio/build/gnuradio-runtime/lib/pmt -I/home/<user>/gnuradio/gnuradio-runtime/lib/../include -I/home/<user>/gnuradio/build/gnuradio-runtime/lib/../include -I/home/<user>/gnuradio/gnuradio-runtime/lib/pmt/../../include -I/home/<user>/gnuradio/build/volk/include -I/home/<user>/gnuradio/volk/include  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -O3 -DNDEBUG   -std=c++11 -o CMakeFiles/pmt_unv.dir/pmt/qa_pmt_unv.cc.o -c /home/<user>/gnuradio/gnuradio-runtime/lib/pmt/qa_pmt_unv.cc
[ 14%] Linking CXX executable pmt_unv
cd /home/<user>/gnuradio/build/gnuradio-runtime/lib && /usr/local/bin/cmake -E cmake_link_script CMakeFiles/pmt_unv.dir/link.txt --verbose=ON
/bin/c++   -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -O3 -DNDEBUG   CMakeFiles/pmt_unv.dir/pmt/qa_pmt_unv.cc.o  -o pmt_unv -Wl,-rpath,/home/<user>/gnuradio/build/gnuradio-runtime/lib:/home/<user>/gnuradio/build/gnuradio-runtime/lib/pmt:/home/<user>/gnuradio/build/volk/lib libgnuradio-runtime.so.3.8tech-preview-381-g27dd99e4 pmt/libgnuradio-pmt.so.3.8tech-preview-381-g27dd99e4 /usr/lib64/libboost_unit_test_framework-mt.so ../../volk/lib/libvolk.so.1.4.1git -ldl /usr/lib64/liborc-0.4.so -lm /usr/lib64/libboost_program_options-mt.so /usr/lib64/libboost_filesystem-mt.so /usr/lib64/libboost_regex-mt.so /usr/lib64/libboost_thread-mt.so /usr/lib64/libboost_atomic-mt.so /usr/lib64/libboost_thread-mt.so /usr/lib64/libboost_atomic-mt.so -lpthread /usr/lib64/libboost_chrono-mt.so /usr/lib64/libboost_system-mt.so /usr/lib64/libboost_date_time-mt.so /usr/lib64/libgmpxx.so /usr/lib64/libgmp.so -lrt 
libgnuradio-runtime.so.3.8tech-preview-381-g27dd99e4: undefined reference to `log4cpp::Category::exists(std::string const&)'
libgnuradio-runtime.so.3.8tech-preview-381-g27dd99e4: undefined reference to `log4cpp::CategoryStream::operator<<(log4cpp::CategoryStream& (*)(log4cpp::CategoryStream&))'
libgnuradio-runtime.so.3.8tech-preview-381-g27dd99e4: undefined reference to `log4cpp::CategoryStream::~CategoryStream()'
libgnuradio-runtime.so.3.8tech-preview-381-g27dd99e4: undefined reference to `log4cpp::FileAppender::FileAppender(std::string const&, std::string const&, bool, unsigned int)'
libgnuradio-runtime.so.3.8tech-preview-381-g27dd99e4: undefined reference to `log4cpp::PropertyConfigurator::configure(std::string const&)'
libgnuradio-runtime.so.3.8tech-preview-381-g27dd99e4: undefined reference to `log4cpp::PatternLayout::PatternLayout()'
libgnuradio-runtime.so.3.8tech-preview-381-g27dd99e4: undefined reference to `log4cpp::Category::getInstance(std::string const&)'
libgnuradio-runtime.so.3.8tech-preview-381-g27dd99e4: undefined reference to `log4cpp::RollingFileAppender::RollingFileAppender(std::string const&, std::string const&, unsigned long, unsigned int, bool, unsigned int)'
libgnuradio-runtime.so.3.8tech-preview-381-g27dd99e4: undefined reference to `log4cpp::eol(log4cpp::CategoryStream&)'
libgnuradio-runtime.so.3.8tech-preview-381-g27dd99e4: undefined reference to `log4cpp::CategoryStream::operator<<(char const*)'
libgnuradio-runtime.so.3.8tech-preview-381-g27dd99e4: undefined reference to `log4cpp::OstreamAppender::OstreamAppender(std::string const&, std::ostream*)'
libgnuradio-runtime.so.3.8tech-preview-381-g27dd99e4: undefined reference to `log4cpp::Category::getCurrentCategories()'
collect2: error: ld returned 1 exit status
make[2]: *** [gnuradio-runtime/lib/pmt_unv] Error 1
make[2]: Leaving directory `/home/<user>/gnuradio/build'
make[1]: *** [gnuradio-runtime/lib/CMakeFiles/pmt_unv.dir/all] Error 2
make[1]: Leaving directory `/home/<user>/gnuradio/build'
make: *** [all] Error 2
[<user>@cosmic-<user> build]$ sudo ldconfig
[<user>@cosmic-<user> build]$ make VERBOSE=ON
/usr/local/bin/cmake -S/home/<user>/gnuradio -B/home/<user>/gnuradio/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/local/bin/cmake -E cmake_progress_start /home/<user>/gnuradio/build/CMakeFiles /home/<user>/gnuradio/build/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: Entering directory `/home/<user>/gnuradio/build'
make -f volk/lib/CMakeFiles/volk_obj.dir/build.make volk/lib/CMakeFiles/volk_obj.dir/depend
make[2]: Entering directory `/home/<user>/gnuradio/build'
cd /home/<user>/gnuradio/build && /usr/local/bin/cmake -E cmake_depends "Unix Makefiles" /home/<user>/gnuradio /home/<user>/gnuradio/volk/lib /home/<user>/gnuradio/build /home/<user>/gnuradio/build/volk/lib /home/<user>/gnuradio/build/volk/lib/CMakeFiles/volk_obj.dir/DependInfo.cmake --color=
make[2]: Leaving directory `/home/<user>/gnuradio/build'
[... cut ...]
libgnuradio-runtime.so.3.8tech-preview-381-g27dd99e4: undefined reference to `log4cpp::OstreamAppender::OstreamAppender(std::string const&, std::ostream*)'
libgnuradio-runtime.so.3.8tech-preview-381-g27dd99e4: undefined reference to `log4cpp::Category::getCurrentCategories()'
collect2: error: ld returned 1 exit status
make[2]: *** [gnuradio-runtime/lib/pmt_unv] Error 1
make[2]: Leaving directory `/home/<user>/gnuradio/build'
make[1]: *** [gnuradio-runtime/lib/CMakeFiles/pmt_unv.dir/all] Error 2
make[1]: Leaving directory `/home/<user>/gnuradio/build'
make: *** [all] Error 2
```

michaelld, 2019-05-29:
Ok so the issue is that there is no "log4cpp" linkage here. Are you trying to build the latest GR GIT master HEAD?  Maybe do "git pull" just to verify ...

If you are, then we need to go back to the "cmake" command & see what it thinks.

pianoman730, 2019-05-29:
I'm cloning with this command `git clone --recursive https://github.com/gnuradio/gnuradio.git`.

Here's the cmake output:
```-- Build type not specified: defaulting to release.
-- Build type set to Release.
-- Extracting version information from git describe...
-- Compiler Version: cc (GCC) 4.8.5 20150623 (Red Hat 4.8.5-36)
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
-- Compiler Flags: /bin/cc:::-O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
/bin/c++:::-O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
-- ADDING PERF COUNTERS
-- Boost version: 1.53.0
-- Found the following Boost libraries:
--   date_time
--   program_options
--   filesystem
--   system
--   regex
--   thread
--   unit_test_framework
--   chrono
--   atomic
-- User set python executable /bin/python3
-- Found PythonLibs: //lib64/libpython3.6m.so (found suitable exact version "3.6.8") 
-- Found PythonLibs: //lib64/libpython3.6m.so (found suitable version "3.6.8", minimum required is "2.7") 
-- Python checking for six - python 2 and 3 compatibility library - found
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.12.
-- 
-- The build system will automatically enable all components.
-- Use -DENABLE_DEFAULT=OFF to disable components by default.
-- 
-- Configuring python-support support...
--   Dependency PYTHONLIBS_FOUND = TRUE
--   Dependency SWIG_FOUND = TRUE
--   Dependency SWIG_VERSION_CHECK = TRUE
--   Dependency SIX_FOUND = TRUE
--   Enabling python-support support.
--   Override with -DENABLE_PYTHON=ON/OFF
-- 
-- Configuring testing-support support...
--   Dependency Boost_FOUND = 1
--   Enabling testing-support support.
--   Override with -DENABLE_TESTING=ON/OFF
-- 
-- Configuring VOLK support...
-- Build type set to Release.
-- Extracting version information from git describe...
-- 
-- Python checking for python >= 2.7
-- Python checking for python >= 2.7 - found
-- 
-- Python checking for mako >= 0.4.2
-- Python checking for mako >= 0.4.2 - found
-- 
-- Python checking for six - python 2 and 3 compatibility library
-- Python checking for six - python 2 and 3 compatibility library - found
-- Boost version: 1.53.0
-- Found the following Boost libraries:
--   filesystem
--   system
-- QA Testing is enabled.
--   Modify using: -DENABLE_TESTING=ON/OFF
-- System profiling is disabled.
--   Modify using: -DENABLE_PROFILING=ON/OFF
-- Compiler name: GNU
-- x86* CPU detected
[... 390 lines cut ...]
WARNING: Target "channels_swig" requests linking to directory "/usr/lib64".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "gnuradio-qtgui" requests linking to directory "/usr/lib64".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "gnuradio-qtgui" requests linking to directory "/usr/lib64".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "display_qt" requests linking to directory "/usr/lib64".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "qtgui_swig" requests linking to directory "/usr/lib64".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "gnuradio-trellis" requests linking to directory "/usr/lib64".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "gnuradio-trellis" requests linking to directory "/usr/lib64".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "trellis_swig0" requests linking to directory "/usr/lib64".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "trellis_swig1" requests linking to directory "/usr/lib64".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "gnuradio-uhd" requests linking to directory "/usr/lib64".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "gnuradio-uhd" requests linking to directory "/usr/lib64".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "tags_demo" requests linking to directory "/usr/lib64".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "uhd_swig" requests linking to directory "/usr/lib64".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "gnuradio-vocoder" requests linking to directory "/usr/lib64".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "gnuradio-vocoder" requests linking to directory "/usr/lib64".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "vocoder_swig" requests linking to directory "/usr/lib64".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "gnuradio-wavelet" requests linking to directory "/usr/lib64".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "gnuradio-wavelet" requests linking to directory "/usr/lib64".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "wavelet_swig" requests linking to directory "/usr/lib64".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "gnuradio-zeromq" requests linking to directory "/usr/lib64".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "gnuradio-zeromq" requests linking to directory "/usr/lib64".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "zeromq_swig" requests linking to directory "/usr/lib64".  Targets may link only to libraries.  CMake is dropping the item.
-- Generating done
-- Build files have been written to: /home/<user>/gnuradio/build
```

pianoman730, 2019-05-29:
@michaelld I installed log4cplus, which is on version 1.1.3 hoping that it would work. I ran the cmake command with the same paths for the log4cpp flags, but when I run make, it errors out saying that there is no such file as log4cpp/Category.hh, which is true, since it's log4cplus instead of log4cpp.

With all of this, the log4cpp version on other distros is 1.1.3. Should I try to build that from source and see how it goes? Otherwise, centos 7 is using 1.1.1.

pianoman730, 2019-05-29:
So I was actually able to get past the linking error for log4cpp by doing what is described [here](https://www.mail-archive.com/osmocom-sdr@lists.osmocom.org/msg01236.html), in that you add
`-DCMAKE_EXE_LINKER_FLAGS=-llog4cpp -DCMAKE_MODULE_LINKER_FLAGS=-llog4cpp -DCMAKE_SHARED_LINKER_FLAGS=-llog4cpp` to the cmake command. 

However, it ran into another error with log4cpp:
```
[ 16%] Linking CXX executable _pmt_swig_doc_tag
cd /home/<user>/gnuradio/build/gnuradio-runtime/swig && /usr/local/bin/cmake -E cmake_link_script CMakeFiles/_pmt_swig_doc_tag.dir/link.txt --verbose=ON
/bin/c++   -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -O3 -DNDEBUG  -llog4cpp CMakeFiles/_pmt_swig_doc_tag.dir/_pmt_swig_doc_tag.cpp.o  -o _pmt_swig_doc_tag 
/usr/lib/gcc/x86_64-redhat-linux/4.8.5/../../../../lib64/liblog4cpp.so: undefined reference to `pthread_key_create'
/usr/lib/gcc/x86_64-redhat-linux/4.8.5/../../../../lib64/liblog4cpp.so: undefined reference to `pthread_getspecific'
/usr/lib/gcc/x86_64-redhat-linux/4.8.5/../../../../lib64/liblog4cpp.so: undefined reference to `pthread_key_delete'
/usr/lib/gcc/x86_64-redhat-linux/4.8.5/../../../../lib64/liblog4cpp.so: undefined reference to `pthread_setspecific'
collect2: error: ld returned 1 exit status
make[2]: *** [gnuradio-runtime/swig/_pmt_swig_doc_tag] Error 1
make[2]: Leaving directory `/home/jacobhadley/gnuradio/build'
make[1]: *** [gnuradio-runtime/swig/CMakeFiles/_pmt_swig_doc_tag.dir/all] Error 2
make[1]: Leaving directory `/home/jacobhadley/gnuradio/build'
make: *** [all] Error 2
```

With the references to log4cpp, I'm assuming that what I did with the cmake command didn't quite fix the issue...

michaelld, 2019-05-29:
OK so adding the `*_LINKER_FLAGS` is just a bandaid & it might work or not temporarily, but it's not a long-term fix. I'm guessing the issue is with the GR-provided log4cpp find routines. Do you mind if we take this to email instead of this issue? I'll keep helping you, and if we find anything truly relevant we can push it somewhere public. My email is "michael.dickens@ettus.com" ... please send me the file from your top-level `build` directory called `CMakeCache.txt` ... it might shed some more light. I'll also try my own build allowing CMake to find the log4cpp (instead of me telling it how to do so, which is my preferred method).

pianoman730, 2019-05-29:
Thanks, I sent you an email.

mwk088, 2019-12-03:
Can we reopen this?  I am getting the same error as desribed here.  Cmake is finding log4cpp, but when I try and make gnuradio, I get another error.  I tried the latest suggests with cmake, but still am getting the same errors.

#40 302.2 [ 29%] Linking CXX executable zeromq_swig_gr_zeromq_swig_ed811
#40 302.4 [ 29%] Linking CXX shared library libgnuradio-pmt-3.7.13.5.so
#40 302.4 Swig source
#40 302.5 /bin/ld: cannot find -llog4cpp
#40 302.5 collect2: error: ld returned 1 exit status
#40 302.5 make[2]: *** [gnuradio-runtime/lib/pmt/CMakeFiles/gnuradio-pmt.dir/build.make:175: gnuradio-runtime/lib/pmt/libgnuradio-pmt-3.7.13.5.so.0.0.0] Error 1
#40 302.5 make[1]: *** [CMakeFiles/Makefile2:790: gnuradio-runtime/lib/pmt/CMakeFiles/gnuradio-pmt.dir/all] Error 2
#40 302.5 make[1]: *** Waiting for unfinished jobs....

CMake
Found LOG4CPP: /opt/xxxx/6.2/lib/liblog4cpp.so
#40 236.9 -- ENABLE_GR_LOG set to ON.
#40 236.9 -- HAVE_LOG4CPP set to True.
#40 236.9 -- LOG4CPP_LIBRARIES set to /opt/xxxx/6.2/lib/liblog4cpp.so.

michaelld, 2019-12-04:
Can you do make VERBOSE=ON and report back the link command that's generating this error? I'm guessing the issue is that Log4CPP is installed outside the CMAKE_INSTALL_PREFIX and thus we need to provide CMake with some library linkage info.

michaelld, 2019-12-04:
@mwk088 can you please open a new issue for this, recreated your entry, provide the info I ask for, and tag me on it? We'll take this up there. This issue is old enough it's really not clear whether it's the same as yours.

mwk088, 2019-12-04:
I may have gotten around the issue by installing the newest version of log4cpp for CentOS 8….it is yet to be determined because of my other linking errors with thrift.  I will open a new ticket if the issue persists.