gnuradio/gnuradio #1046: make install doesn't work

maxbit89, 2016-10-03:
I am using ubuntu 16.04 LTS, i installed gnuradio with the folowing commands:
mkdir build && cd build
cmake ..
make -j 4
make test -j 4
sudo make install

it installs gnuradio to /usr/local/...
but i can't start gnuradio-companion

jmcorgan, 2016-10-03:
You also need to run 'sudo ldconfig'.  Also, make sure that GRC was actually included in the build (you should see that in the cmake output.)

maxbit89, 2016-10-03:
That was fast :+1: 
i used this command  to clone the ropo: git clone --recursive http://git.gnuradio.org/git/gnuradio.git
and yes i didn't find any thing about gnuradio-companion

jmcorgan, 2016-10-03:
If you are going to do a manual source build, then you'll need to make sure you have all the dependencies correctly installed.  I suggest you look at using the PyBOMBS utility to automate that for you, or even use the prepackaged 3.7.9.1 that is in the Ubuntu repository.

In any case, we prefer using the mailing list for support issues, and here for reporting verifiable bugs or making feature requests.

mbr0wn, 2016-10-03:
Since this doesn't look like an actual GNU Radio issue, let's close it. @maxbit89 as @jmcorgan says, please prefer the mailing list for these kinds of issues.

closed 2016-10-03 by mbr0wn