gnuradio/gnuradio #1047: User reports segfault  running g-perf-monitorx

balister, 2016-10-04:
Got further this time:
root@b4860:~# gr-perf-monitorx
Unable to open locale C in ICU, error code 3 Could not connect to 
ControlPort endpoint at None:9090.

Exception socket.error: (32, 'Broken pipe') in <bound method 
ThriftRadioClient.__del__ of 
<gnuradio.ctrlport.RPCConnectionThrift.ThriftRadioClient instance 
at
0x134986c8>> ignored root@b4860:~#

balister, 2016-10-04:
Tracked down to:

On 10/04/2016 03:07 PM, kyle.unice@L-3com.com wrote:
> Philip,
>
> Turns out the BSS initialize was initializing these global variables in the wrong order. These two patches fixed it.
>
> K-

We need to review the attached patches.

--- a/gnuradio-runtime/lib/controlport/rpcmanager.cc	2016-10-04 15:04:30.983817995 -0700
+++ b/gnuradio-runtime/lib/controlport/rpcmanager.cc	2016-10-04 15:01:23.804881925 -0700
@@ -24,10 +24,6 @@
 #include <iostream>
 #include <stdexcept>
 
-bool rpcmanager::booter_registered(false);
-bool rpcmanager::aggregator_registered(false);
-std::auto_ptr<rpcserver_booter_base> rpcmanager::boot(0);
-std::auto_ptr<rpcserver_booter_aggregator> rpcmanager::aggregator(0);
 
 rpcmanager::rpcmanager() {;}
 
and

--- a/gnuradio-runtime/lib/controlport/rpcserver_selector.cc	2016-10-04 15:04:30.983817995 -0700
+++ b/gnuradio-runtime/lib/controlport/rpcserver_selector.cc	2016-10-04 15:01:34.044384104 -0700
@@ -25,6 +25,10 @@
 #include <gnuradio/rpcserver_selector.h>
 
 bool rpcmanager::make_aggregator(false);
+bool rpcmanager::booter_registered(false);
+bool rpcmanager::aggregator_registered(false);
+std::auto_ptr<rpcserver_booter_base> rpcmanager::boot(0);
+std::auto_ptr<rpcserver_booter_aggregator> rpcmanager::aggregator(0);
 
 #ifdef GR_RPCSERVER_ENABLED
 rpcmanager manager_instance;

Also, github refuses to let me attach .patch files. Feel the anger.

jmcorgan, 2016-10-23:
Fixed in #1065

closed 2016-10-23 by jmcorgan